openrazer/openrazer #824: Daemon not recognizing Razer Blade 15 Advanced 2019 (Mercury)

baa066, 2019-04-29:
Hello,

Coming from Manjaro on a Razer Blade 15 2019 Mercury Edition. I installed openrazer through pacaur and have also downloaded Polychromatic for controls. Unfortunately, Polychromatic reports "A Razer product is connected to your computer, but the daemon didn't detect it. It is possible that the product is not bound to the driver properly or is not yet supported."

Any troubleshooting pointers?

Thanks!

I believe I have a compatible device:
lsusb | grep '1532:'
Bus 001 Device 003: ID 1532:023a Razer USA, Ltd

z3ntu, 2019-04-29:
What's the output of the command `sudo dkms status` ? You might be missing your kernel headers.

baa066, 2019-04-29:
sudo dkms status returns no output.

z3ntu, 2019-04-29:
What about `sudo modprobe razerkbd` ?

zephyrfalcon, 2019-05-10:
FWIW, I have the same problem. Also, `sudo dkms status` shows nothing at all for me.

z3ntu, 2019-05-22:
@zephyrfalcon Install your kernel headers please, then you can run 
`sudo dkms install openrazer-driver/2.5.0`

zephyrfalcon, 2019-05-23:
Thanks @z3ntu. I now get the following output:

```
$ sudo dkms status
openrazer-driver, 2.5.0, 4.19.42-1-MANJARO, x86_64: installed
```

I am still getting the same error as @baa066 when I run Polychromatic. :(

z3ntu, 2019-05-25:
@zephyrfalcon Did you try rebooting?

zephyrfalcon, 2019-05-25:
@z3ntu I thought I did... maybe not... it seems to work now, in the sense that the daemon starts properly and finds the device. Polychromatic has a problem with selecting the keys, but that is probably not related to OpenRazer. Thanks for your help.

lah7, 2019-05-25:
> Polychromatic has a problem with selecting the keys, but that is probably not related to OpenRazer.

You might want to try the patch in https://github.com/polychromatic/polychromatic/pull/144, which modified the keyboard graphic to work for the Blade Stealth (Late 2016) - it might work for your model too:

    wget https://raw.githubusercontent.com/polychromatic/polychromatic/8607f48d6776ec3d3c84150b7b2506775590b890/data/img/blackwidow-chroma-keyboard-layout.svg
    sudo mv blackwidow-chroma-keyboard-layout.svg /usr/share/polychromatic/mapping/

closed 2019-10-22 by stale[bot]