openrazer/openrazer #825: audio controls for keyboard

codeman101, 2019-05-02:
### Distribution

Manjaro KDE

### Support

- [x ] Have you tried rebooting?
- [x ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

I noticed what appears to be key-mappings to the audio controls on the keyboard but they don't work for me when I use spotify for example and I use Razergenie as the front end which only deals with lighting.

z3ntu, 2019-05-02:
What device is this about?

codeman101, 2019-05-02:
@z3ntu  Sorry it's the Blackwidow X Chroma

JohnMarzulli, 2019-05-11:
Occurs with Pop!_OS 19.04 and Huntsman Elite.

Stopping the Daemon restores the volume controls, but deactivates ALT+TAB, "Windows Button", and more.

Restarting the Daemon restores the Windows button, but disables audio controls.

andrewschott, 2019-05-19:
Not just PopOS, but also Fedora and OpenSUSE

z3ntu, 2019-07-13:
Is this the equivalent of #803 for the Blackwidow X Chroma?

codeman101, 2019-07-13:
@z3ntu It doesn't sound like it based on the title of that issue and plus that issue is closed. This is on going.

z3ntu, 2019-07-13:
At least the comment from @andrewschott 
> From digging around with xev, it looks like the udev is seeing the wheel as a mouse wheel, as it mimicks perfectly my mouse wheel.

sounds a lot like #803 (which was btw closed 3 hours ago as I pushed a fix for that device)

@codeman101 
Can you explain the issue you're seeing in more detail? I don't understand your issue from the top message.

codeman101, 2019-07-13:
@z3ntu a very simple example of the issue is if you use Spotify you won't be able to use the volume controls or audio controls at all to control the program. I know this is not an issue with Spotify because it works fine on my laptop that runs the same Linux distribution just a different desktop environment.

z3ntu, 2019-07-13:
So Fn+F2 doesn't produce the "Decrease volume" button, am I understanding you correctly? That sounds to be related to #872 then (or see #873 for a bit more information)

codeman101, 2019-07-13:
@z3ntu that's correct. Those are definitely more like this issue although to be honest I never realized pressing the FN key was required but that definitely sounds like the same thing. Also to be honest I'm surprised it has yet to be solved because as far as I can tell it's the only hardware issue with the driver that I am aware of at least in terms of importance. Is there any chance the issue that this thread and the otherd related to it are talking about could be given higher priority?

andrewschott, 2019-08-01:
I can say that on my Huntsman Elite, the audio wheel now works perfectly as does the multimedia controls.  This is as of 2.6.0 release.  Thanks gang!

codeman101, 2019-12-11:
Someone should reopen this. Closing it only causes it to be re-created especially given the reason it was closed.

N3TC4T, 2020-07-11:
You can use https://wiki.archlinux.org/index.php/Xbindkeys

here is my config controlling spotify and volume 

Note: key maps can be different base on your keyboard model
```

"amixer -q sset Master 5%+"
    m:0x10 + c:123
    Mod2 + XF86AudioRaiseVolume

"amixer -q sset Master 5%-"
    m:0x10 + c:122
    Mod2 + XF86AudioLowerVolume

"dbus-send --print-reply --dest=org.mpris.MediaPlayer2.spotify /org/mpris/MediaPlayer2 org.mpris.MediaPlayer2.Player.PlayPause"
    m:0x10 + c:172
    Mod2 + XF86AudioPlay

"dbus-send --print-reply --dest=org.mpris.MediaPlayer2.spotify /org/mpris/MediaPlayer2 org.mpris.MediaPlayer2.Player.Previous"
    m:0x10 + c:173
    Mod2 + XF86AudioPrev

"dbus-send --print-reply --dest=org.mpris.MediaPlayer2.spotify /org/mpris/MediaPlayer2 org.mpris.MediaPlayer2.Player.Next"
    m:0x10 + c:171
    Mod2 + XF86AudioNext

"amixer -q sset Master 1+ toggle"
    m:0x10 + c:121
    Mod2 + XF86AudioMute

```

closed 2026-10-30 by codeman101