openrazer/openrazer #826: Unable to load module razerkbd

ctrl-alt-d, 2019-05-05:
### Distribution

Fedora 29

### Support

- [x] Have you tried rebooting? Yes
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page? Yes

I'm not able to load `razerkbd` module:

```
[dani@localhost ~]$ modprobe razerkbd
modprobe: FATAL: Module razerkbd not found in directory /lib/modules/5.0.10-200.fc29.x86_64
```

I have updated all packages but not found `razerkbd` for my kernel:

```
[dani@localhost ~]$ locate razerkbd
/usr/lib/modules/4.15.17-300.fc27.x86_64/extra/razerkbd.ko.xz
/usr/lib/modules/4.16.11-300.fc28.x86_64/extra/razerkbd.ko.xz
/usr/lib/modules/4.16.12-300.fc28.x86_64/extra/razerkbd.ko.xz
/usr/lib/modules/4.16.13-300.fc28.x86_64/extra/razerkbd.ko.xz
/usr/lib/modules/4.16.14-300.fc28.x86_64/extra/razerkbd.ko.xz
/usr/lib/modules/4.16.16-300.fc28.x86_64/extra/razerkbd.ko.xz
/usr/lib/modules/4.16.5-200.fc27.x86_64/extra/razerkbd.ko.xz
/usr/lib/modules/4.16.5-300.fc28.x86_64/extra/razerkbd.ko.xz
/usr/lib/modules/4.16.8-300.fc28.x86_64/extra/razerkbd.ko.xz
/usr/lib/modules/4.17.2-200.fc28.x86_64/extra/razerkbd.ko.xz
/usr/lib/modules/4.17.3-200.fc28.x86_64/extra/razerkbd.ko.xz
/usr/src/openrazer-driver-2.5.0/driver/razerkbd_driver.c
/usr/src/openrazer-driver-2.5.0/driver/razerkbd_driver.h
/var/lib/dkms/openrazer-driver/1.0.0/4.15.17-300.fc27.x86_64/x86_64/module/razerkbd.ko.xz
/var/lib/dkms/openrazer-driver/1.0.0/4.16.11-300.fc28.x86_64/x86_64/module/razerkbd.ko.xz
/var/lib/dkms/openrazer-driver/1.0.0/4.16.12-300.fc28.x86_64/x86_64/module/razerkbd.ko.xz
/var/lib/dkms/openrazer-driver/1.0.0/4.16.13-300.fc28.x86_64/x86_64/module/razerkbd.ko.xz
/var/lib/dkms/openrazer-driver/1.0.0/4.16.14-300.fc28.x86_64/x86_64/module/razerkbd.ko.xz
/var/lib/dkms/openrazer-driver/1.0.0/4.16.16-300.fc28.x86_64/x86_64/module/razerkbd.ko.xz
/var/lib/dkms/openrazer-driver/1.0.0/4.16.5-200.fc27.x86_64/x86_64/module/razerkbd.ko.xz
/var/lib/dkms/openrazer-driver/1.0.0/4.16.5-300.fc28.x86_64/x86_64/module/razerkbd.ko.xz
/var/lib/dkms/openrazer-driver/1.0.0/4.16.8-300.fc28.x86_64/x86_64/module/razerkbd.ko.xz
/var/lib/dkms/openrazer-driver/1.0.0/4.17.2-200.fc28.x86_64/x86_64/module/razerkbd.ko.xz
/var/lib/dkms/openrazer-driver/1.0.0/4.17.3-200.fc28.x86_64/x86_64/module/razerkbd.ko.xz

```

Thanks about to keep up to date all these drivers.

z3ntu, 2019-05-05:
Please run `sudo dkms install openrazer-driver/2.5.0` . And make sure you have the kernel-devel package installed

ctrl-alt-d, 2019-05-05:
Hi @z3ntu ,

thanks about your fast reply and your help!

After run your command driver loads:

```
[dani@localhost ~]$ tree /sys/bus/hid/drivers
/sys/bus/hid/drivers
├── hid-generic
│   ├── 0003:1D57:AD17.0010 -> .
...
├── razerkbd
│   ├── 0003:1532:022D.0007 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:1532:022D.0007
│   ├── 0003:1532:022D.0008 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.1/0003:1532:022D.0008
│   ├── 0003:1532:022D.000A -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.2/0003:1532:022D.000A
│   ├── bind
│   ├── module -> ../../../../module/razerkbd
│   ├── new_id
│   ├── uevent
│   └── unbind

```

Sorry about noob question but I didn't found info about this command on web :(  

Now, driver is load, however, polychromatic don't recognize device:

[image]

Any suggestion?

z3ntu, 2019-05-05:
Please try rebooting now.

d0tfile, 2019-05-05:
How did you install openrazer? I don't see `razerkbd` for you kernel (`5.0.10`)

ctrl-alt-d, 2019-05-05:
Hi @d0tfile, I install openrazer following open-razer instructions:

[image]

Hi @z3ntu, thanks again, it works like a charm:

[image]

Congrats to open-razor community, great work!

closed 2019-05-05 by ctrl-alt-d

z3ntu, 2019-05-05:
No problem!