openrazer/openrazer #827: Manjaro daemon not starting

Adam-B-Davenport, 2019-05-06:
### Distribution
Manjaro: Kernel 5.0.7

### Support

- [x] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

I can not get the daemon to start, below is the output of openrazer-daemon -Fv.

Starting daemon.
Traceback (most recent call last):
  File "/usr/lib/python3.7/site-packages/daemonize.py", line 248, in start
    self.action(*privileged_action_result)
  File "/usr/bin/openrazer-daemon", line 125, in run_daemon
    test_dir=args.test_dir)
  File "/usr/lib/python3.7/site-packages/openrazer_daemon/daemon.py", line 91, in __init__
    DBusService.__init__(self, self.BUS_NAME, '/org/razer')
  File "/usr/lib/python3.7/site-packages/openrazer_daemon/dbus_services/service.py", line 83, in __init__
    bus_object = dbus.service.BusName(bus_name, bus=dbus.SessionBus())
  File "/usr/lib/python3.7/site-packages/dbus/_dbus.py", line 211, in __new__
    mainloop=mainloop)
  File "/usr/lib/python3.7/site-packages/dbus/_dbus.py", line 100, in __new__
    bus = BusConnection.__new__(subclass, bus_type, mainloop=mainloop)
  File "/usr/lib/python3.7/site-packages/dbus/bus.py", line 122, in __new__
    bus = cls._new_for_bus(address_or_type, mainloop=mainloop)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoServer: Failed to connect to socket /tmp/dbus-JGdor9V6OA: Connection refused

Stopping daemon.

z3ntu, 2019-05-06:
It looks like your dbus session bus setup is broken for some reason. Maybe ask in some Manjaro forum how you can fix it.

Can you run `dbus-test-tool echo` ?

Adam-B-Davenport, 2019-05-09:
The issue ends up having to deal with BSPWM and dbus rather than Manjaro and dbus. The tests run fine but I seem to have issues running a lot of applications that rely upon dbus. I had no trouble with the application in any other desktop environment.

closed 2019-05-09 by Adam-B-Davenport