openrazer/openrazer #828: Fedora 25 Repo (unsupported now?)

flamusdiu, 2019-05-07:
### Distribution

Qubes OS (Fedora 25)

### Support

- [ ] Have you tried rebooting?
- [X] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Is there any location for the Fedora 25 repo? I am running Qubes OS which happens to use Fedora 25 at the moment on my Razer Blade Stealth (mid 2017) and I ran across these drivers which I was hoping would fixed sleep/hibernate on the laptop.

z3ntu, 2019-05-07:
Fedora 25 is end of life for about 1.5 years now. The latest Fedora release is supported though.  If you really want to use the old version you should be able to build the package yourself using the spec file at https://github.com/openrazer/OBS-packaging/blob/master/openrazer.spec

closed 2019-05-07 by z3ntu

flamusdiu, 2019-05-11:
I would love to use the latest release but Qubes OS is built using Fedora 25 so I probably should use that. Thanks for the information and I'll work on getting that installed.

flamusdiu, 2019-05-11:
It fails building due to missing object files (not built before hand?). This is after performing the command `sudo dkms openrazer-driver/2.5.0`

Logfile: [make.log](https://github.com/openrazer/openrazer/files/3169048/make.log)

z3ntu, 2019-05-12:
All I can say without actually trying it myself is, that we didn't change anything (significant?) in terms of building everything of OpenRazer since Fedora 25 was supported and it worked fine back then.

flamusdiu, 2019-05-12:
I tried with a VM with Fedora 29 with same problem when I follow these steps: https://openrazer.github.io/#fedora

After those, I then do `sudo dkms build openrazer-driver/2.5.0` which yields me the same error.

z3ntu, 2019-05-12:
The same also with `sudo dkms install openrazer-driver/2.5.0` ?

flamusdiu, 2019-05-12:
Yes. I even attempted other release versions but they seem to all do it as well. There is something missing (or that I am doing wrong).

flamusdiu, 2019-05-12:
When I finish running the two commands, it doesn't seem to build the driver but setups dkms to build it. So run the build manually; i would guess that should work.

z3ntu, 2019-05-13:
[image]
That's a screenshot from a Fedora 29 VM where I ran `sudo dnf install openrazer-meta kernel-devel` and the build worked fine. What exact commands are you running?

flamusdiu, 2019-05-15:
hrms, let me see then; yours is using the 5.0 kernel while mine is still 4.14 (same as dom0) so let get that updated in the VM and see what happens. Thanks for testing that out.