openrazer/openrazer #830: Device not binding to driver

arty-hlr, 2019-05-08:
### Distribution

Parrot 4.19

### Support

- [x] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Hi,

I'm trying to make polychromatic work with my Razer Blade Stealth 2019, and somehow I can't seem to bind the device to the razerkbd driver...

I followed the instructions from the [wiki](https://github.com/openrazer/openrazer/wiki/Setting-up-the-keyboard-driver) and I'm stuck at binding the device to the driver, here is the output of `echo -n 0003:1532:0239.0004 > /sys/bus/hid/drivers/razerkbd/bind` (run as root, no sudo):

```echo: write error: no such device```

I did unbind the device from `hid-generic`, but for some reason it doesn't find the device to bind it...

Here is the output of `ls -l /sys/bus/hid/devices/`:
```total 0
lrwxrwxrwx 1 root root 0 May  8 19:11 0003:1532:0239.0002 -> ../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:1532:0239.0002
lrwxrwxrwx 1 root root 0 May  8 19:11 0003:1532:0239.0003 -> ../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.1/0003:1532:0239.0003
lrwxrwxrwx 1 root root 0 May  8 19:11 0003:1532:0239.0004 -> ../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.2/0003:1532:0239.0004
lrwxrwxrwx 1 root root 0 May  8 19:11 0018:06CB:CDA3.0001 -> ../../../devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-2/i2c-1A581000:00/0018:06CB:CDA3.0001
```
There doesn't seem to be anything related to this in `dmesg`, and I can't find a working solution in other issues, I find the same problem [here](https://github.com/openrazer/openrazer/issues/122) but without a description of how he solved it though... I did `sudo dkms remove razer_chroma_driver/1.0.0 --all && sudo dkms install razer_chroma_driver/1.0.0` and even rebooted, to no avail...

Did anyone run into this as well or does anyone have some hints about things to try?

Thanks in advance! :)

z3ntu, 2019-05-08:
Can you please run `sudo dkms install openrazer-driver/2.5.0` and `sudo modprobe razerkbd` ?

arty-hlr, 2019-05-08:
@z3ntu `sudo dkms install openrazer-driver/2.5.0` gives me an error:
```Error! Could not find module source directory.
Directory: /usr/src/openrazer-driver-2.5.0 does not exist.```
and `sudo modprobe razerkbd` doesn't give me any output, but I guess that's because it worked.

z3ntu, 2019-05-08:
Please make sure you have the latest version (2.5.0) installed!

arty-hlr, 2019-05-08:
Well I tried, but running the dkms install doesn't work, installing the .deb package from [here](https://software.opensuse.org/download.html?project=hardware%3Arazer&package=openrazer-meta) doesn't work as it needs openrazer-driver-dkms version 2.5.0, which I cannot find anywhere, when I do `apt install openrazer-driver-dkms/2.5.0` it tells me that there isn't such a release and that version 2.4.0 is the newest version.

Am I missing something here?

z3ntu, 2019-05-08:
Please add the repository (section "Add repository and install manually") - don't just download and install .deb packages directly.

arty-hlr, 2019-05-08:
I did ^^ The repository was added, and it shows the latest version as being 2.4.0.

arty-hlr, 2019-05-08:
I do have the latest stable version though, 2.4.0, as 2.5.0 as marked as experimental, the problem doesn't seem to come from the version.

z3ntu, 2019-05-08:
2.4.0 isn't even in the OBS repo anymore - and I never marked any release as "experimental". Where do you see that?

z3ntu, 2019-05-08:
Ugh that's the version provided by Debian itself...

arty-hlr, 2019-05-08:
[Here](https://tracker.debian.org/pkg/openrazer) under versions.

arty-hlr, 2019-05-08:
Ok, where should I get version 2.5.0 then?

z3ntu, 2019-05-08:
Install the version from our repos, otherwise bug the Debian maintainer about your problems.

For our version follow https://openrazer.github.io/#debian

arty-hlr, 2019-05-08:
I just did it again, adding your repository, updating, and installing openrazer-meta, and it still installs version 2.4.0... And the .deb package doesn't work as it depends on openrazer-daemon and openrazer-driver-dkms version 2.5.0...

Any idea on the binding problem though?

z3ntu, 2019-05-08:
If you have 2.4.0 installed, then run `sudo dkms install openrazer-driver/2.4.0` and `sudo modprobe razerkbd`

arty-hlr, 2019-05-08:
That first command gives me that: `Module openrazer-driver/2.4.0 already installed on kernel 4.19.0-parrot4-28t-amd64/x86_64` and the second one nothing.

arty-hlr, 2019-05-08:
I managed to install 2.5.0 by buiilding the packages from source though, so that's that.

arty-hlr, 2019-05-08:
Okay, just rebooted, now the keyboard doesn't work anymore, so I can't even login, except for the F8 and F9 keys which put more or less light under the keyboard...

arty-hlr, 2019-05-08:
I managed to log in with VNC and bind the three devices to `hid-generic` again, which works, but the same thing with `razerkbd` doesn't work and says `No such device`.

z3ntu, 2019-05-08:
Don't try to bind it manually, we have a script that's being called by udev to bind it.
You can get the logs for that with the command `sudo journalctl -b | grep -i razer_mount`

arty-hlr, 2019-05-08:
Ok, but that's what it says on the wiki though. Where is that script?

Here is the output:
```
May 08 23:37:02 parrot razer_mount[851]: Driver razerkbd
May 08 23:37:02 parrot razer_mount[857]: Driver razerkbd
May 08 23:37:02 parrot razer_mount[865]: Device_ID 0003:1532:0239.0004
May 08 23:37:02 parrot razer_mount[869]: Device_ID 0003:1532:0239.0003
May 08 23:37:02 parrot razer_mount[873]: Driver razerkbd
May 08 23:37:02 parrot razer_mount[874]: Modprobing razerkbd
May 08 23:37:02 parrot razer_mount[878]: Device_ID 0003:1532:0239.0002
May 08 23:37:02 parrot razer_mount[881]: Modprobing razerkbd
May 08 23:37:02 parrot razer_mount[875]: Modprobing razerkbd
May 08 23:37:02 parrot razer_mount[937]: Modprobed razerkbd
May 08 23:37:02 parrot razer_mount[939]: Modprobed razerkbd
May 08 23:37:02 parrot razer_mount[943]: Unbinding 0003:1532:0239.0002 from hid-generic
May 08 23:37:02 parrot razer_mount[944]: Unbinding 0003:1532:0239.0004 from hid-generic
May 08 23:37:02 parrot razer_mount[940]: Modprobed razerkbd
May 08 23:37:02 parrot razer_mount[952]: Unbinding 0003:1532:0239.0003 from hid-generic
May 08 23:37:05 parrot razer_mount[1379]: Binding 0003:1532:0239.0002 to razerkbd
May 08 23:37:05 parrot systemd-udevd[827]: Process 'razer_mount razerkbd 0003:1532:0239.0002' failed with exit code 1.
May 08 23:37:05 parrot razer_mount[1405]: Binding 0003:1532:0239.0004 to razerkbd
May 08 23:37:05 parrot systemd-udevd[801]: Process 'razer_mount razerkbd 0003:1532:0239.0004' failed with exit code 1.
May 08 23:37:06 parrot razer_mount[1457]: Binding 0003:1532:0239.0003 to razerkbd
May 08 23:37:06 parrot systemd-udevd[794]: Process 'razer_mount razerkbd 0003:1532:0239.0003' failed with exit code 1.
```

z3ntu, 2019-05-08:
> Ok, but that's what it says on the wiki though.

I thought the warning at the top of the page was clear enough.

> Where is that script?

/usr/lib/udev/razer_mount or on Debian I think without the /usr

> Process 'razer_mount razerkbd 0003:1532:0239.0002' failed with exit code 1.

Yep, that's the problem. Normally the script only fails when it can't load razerkbd but it can on your system. No idea why isn't able to bind the device again. If you're still in the same session, you can try to run the script manually with the parameters given (or just echo to the razerkbd `bind` file in sysfs with the 0003:1532:0239.0002` argument)

arty-hlr, 2019-05-08:
Yes, it gets the same I/O error:
```
sudo /lib/udev/razer_mount razerkbd 0003:1532:0239.0004
Driver razerkbd
Device_ID 0003:1532:0239.0004
Unbinding 0003:1532:0239.0004 from hid-generic
Binding 0003:1532:0239.0004 to razerkbd
/lib/udev/razer_mount: 50: printf: printf: I/O error
```
But, running it as root and not with sudo doesn't give that error! The only thing is that it doesn't appear in the directory then.

arty-hlr, 2019-05-08:
Rebooting unbinds everything again though.

arty-hlr, 2019-05-08:
I have removed the kernel module, uninstalled purged and reinstalled openrazer-meta, run the udev script, and the device is still not in the razerkbd directory :/ Polychromatic doesn't recognize the keyboard either.

arty-hlr, 2019-05-08:
And even after removing the kernel module, the devices are not bound to hid-generic anymore on reboot, so in the end trying to make the driver work didn't work, but broke something...

z3ntu, 2019-05-09:
Do you maybe still have files left over from the openrazer installation? Especially the udev rule and the razer_mount script?

arty-hlr, 2019-05-09:
I reinstalled operazer after solving the binding problem for the keyboard, so yes I still have all the files left. I tried the udev script again, but nothing changed unfortunately, it still says `No such device` again.

closed 2019-10-06 by stale[bot]