openrazer/openrazer #832: Razer keyboard not recognized anymore

arty-hlr, 2019-05-09:
### Distribution

Parrot 4.19

### Support

- [x] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

So, after trying without success so far to bind the keyboard devices to the razerkbd driver, I ended up with these devices not even binding to hid-generic on start-up, so the keyboard doesn't work at start-up (except for the F10 and F11 keys for the keyboard lighting...).

I wrote a script which run as root binds the devices to the hid-generic driver (that works), and I've tried a few ways to have that script run at start-up or reboot (created a systemctl service which fails until I restart it, tried a root cron job with `@reboot`, doesn't work either).

Was anyone in that position with a Razer keyboard at some point? I'd be glad for even a hint or a way to get things how they were before!

Thanks in advance :)

z3ntu, 2019-05-09:
Do you maybe still have files left over from the openrazer installation? Especially the udev rule and the razer_mount script?

z3ntu, 2019-05-09:
Let's continue in #830, no need to open a separate issue

closed 2019-05-09 by z3ntu