openrazer/openrazer #834: Support for Razer Blade 15 (Mid-2019) Mercury White

jonjohnsontc, 2019-05-12:
### Device Info

* Name: <See under device>
* Product Number: <See under device>
* Model Number: <See under device>

### Device Information
```
Please post the output of
    lsusb -d 1532: -v

Bus 001 Device 006: ID 1532:0245 Razer USA, Ltd 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0245 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
```

```
Please post the output of
    ls -lah /dev/input/by-id/

total 0
drwxr-xr-x 2 root root 220 May 12 13:51 .
drwxr-xr-x 4 root root 680 May 12 13:51 ..
lrwxrwxrwx 1 root root  10 May 12 13:33 usb-Azurewave_Integrated_Camera_200901010001-event-if00 -> ../event12
lrwxrwxrwx 1 root root  10 May 12 13:33 usb-Azurewave_Integrated_Camera_200901010001-event-if02 -> ../event13
lrwxrwxrwx 1 root root  10 May 12 13:33 usb-Logitech_USB_Receiver-if02-event-mouse -> ../event11
lrwxrwxrwx 1 root root   9 May 12 13:33 usb-Logitech_USB_Receiver-if02-mouse -> ../mouse1
lrwxrwxrwx 1 root root   9 May 12 13:33 usb-Razer_Razer_Blade-event-if01 -> ../event9
lrwxrwxrwx 1 root root   9 May 12 13:33 usb-Razer_Razer_Blade-event-kbd -> ../event5
lrwxrwxrwx 1 root root   9 May 12 13:33 usb-Razer_Razer_Blade-if01-event-kbd -> ../event6
lrwxrwxrwx 1 root root  10 May 12 13:33 usb-Razer_Razer_Blade-if02-event-mouse -> ../event10
lrwxrwxrwx 1 root root   9 May 12 13:33 usb-Razer_Razer_Blade-if02-mouse -> ../mouse0

```

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on Razer Synapse>

mgruppi, 2019-05-22:
Hello. I've modified to drivers to support this mercury white model. Care to test and let us know if it works?

jonjohnsontc, 2019-05-23:
@mgruppi yes I'd be happy to. Is it available in the stable build or should I download the daily instead?

mgruppi, 2019-05-23:
It hasn't been merged yet so you will have to install it manually or wait until it gets merged into daily.

If you don't wanna wait, I recommend that, first, you install the daily build from ppa (if you're on Ubuntu or similar). This will give you the current daily build, which doesn't yet support the Mid 2019 Blade.

Then, clone/download this `https://github.com/mgruppi/openrazer.git ` into your system and run `make` from within the cloned directory. This will compile the keyboard drivers that support your system. If it works, you should see the file `openrazer/driver/razerkbd.ko`.

Now we need to copy the newly compiled driver to the right location. Run `locate razerkbd.ko`, on Ubuntu you may some something like `/lib/modules/4.18.0-XX-generic/updates/dkms/razerkbd.ko`. You can a backup of that file. Then replace it with the newly compiled razerkbd.ko.

We need to update one more file, copy from the cloned directory `openrazer/daemon/openrazer_daemon/hardware/keyboards.py` to `/usr/lib/python3/dist-packages/openrazer_daemon/hardware/keyboards.py`.

Reboot and test it out with one of the frontends or with the Python API. Hope it works!

jonjohnsontc, 2019-05-26:
Thanks so much for the detailed instructions! 

Unfortunately, Polychromatic doesn't detect my device, and when I run the following to check for devices using the Python library, I see an empty list

```
import openrazer.client
devman = openrazer.client.DeviceManager()
devices = devman.devices
devices
```

I had tried installing the stable build a little while back. Do you think that's what's causing the issue? Would I need to prune any previous drivers?

Thanks again,

mgruppi, 2019-05-26:
The previous install should not be a problem. Installing it manually has a bunch of tricky parts.
I forgot about the udev. You must copy `openrazer/install_files/udev/99-razer.rules `to your system too, locate it, probably in `/lib/udev/rules.d/`, then replace it with the one from the cloned directory.

jonjohnsontc, 2019-05-27:
Thanks so much! I did that as well and rebooted, but still no luck unfortunately. Do you know if I have to do any additional steps, by chance?

mgruppi, 2019-06-03:
I'm not sure what else it's missing. When the PR gets merged you can easily install it directly from the daily build repo.

jonjohnsontc, 2019-06-04:
Sounds great. Thanks so much for your help.

mgruppi, 2019-06-16:
Good news, it's been merged. You can install it via the daily repo probably by tomorrow morning.

z3ntu, 2019-06-17:
Fixed by #835

closed 2019-06-17 by z3ntu