openrazer/openrazer #836: Support for [device name here]

amase4, 2019-05-14:
### Device Info

* Name: Kraken Tournament Edition
* Product Number: RZ04-02051000
* Model Number: RZ04-0205

### Device Information
```
Please post the output of
    lsusb -d 1532: -v
```Bus 001 Device 005: ID 1532:0520 Razer USA, Ltd 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0520 
  bcdDevice            0.88
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0198
    bNumInterfaces          6
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass         1 Audio
      bInterfaceSubClass      1 Control Device
      bInterfaceProtocol      0 
      iInterface              5 
      AudioControl Interface Descriptor:
        bLength                10
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdADC               1.00
        wTotalLength       0x0048
        bInCollection           2
        baInterfaceNr(0)        1
        baInterfaceNr(1)        2
      AudioControl Interface Descriptor:
        bLength                12
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID             1
        wTerminalType      0x0402 Headset
        bAssocTerminal          6
        bNrChannels             2
        wChannelConfig     0x0003
          Left Front (L)
          Right Front (R)
        iChannelNames           0 
        iTerminal               0 
      AudioControl Interface Descriptor:
[... 491 lines cut ...]

```
Please post the output of
    ls -lah /dev/input/by-id/
```total 0
drwxr-xr-x 2 root root 240 May 13 20:17 .
drwxr-xr-x 4 root root 580 May 13 20:17 ..
lrwxrwxrwx 1 root root  10 May 13 20:17 usb-Blue_Microphones_Yeti_Stereo_Microphone_REV8-event-if03 -> ../event11
lrwxrwxrwx 1 root root  10 May 13 20:17 usb-Logitech_Gaming_Mouse_G502_1576375A3030-event-if01 -> ../event16
lrwxrwxrwx 1 root root   9 May 13 20:17 usb-Logitech_Gaming_Mouse_G502_1576375A3030-event-mouse -> ../event5
lrwxrwxrwx 1 root root  10 May 13 20:17 usb-Logitech_Gaming_Mouse_G502_1576375A3030-if01-event-kbd -> ../event15
lrwxrwxrwx 1 root root   9 May 13 20:17 usb-Logitech_Gaming_Mouse_G502_1576375A3030-mouse -> ../mouse1
lrwxrwxrwx 1 root root  10 May 13 20:17 usb-Razer_Razer_BlackWidow_Tournament_Edition_Chroma-event-kbd -> ../event13
lrwxrwxrwx 1 root root  10 May 13 20:17 usb-Razer_Razer_BlackWidow_Tournament_Edition_Chroma-if01-event-kbd -> ../event21
lrwxrwxrwx 1 root root   9 May 13 20:17 usb-Razer_Razer_BlackWidow_Tournament_Edition_Chroma-if02-event-mouse -> ../event8
lrwxrwxrwx 1 root root   9 May 13 20:17 usb-Razer_Razer_BlackWidow_Tournament_Edition_Chroma-if02-mouse -> ../mouse0
lrwxrwxrwx 1 root root  10 May 13 20:17 usb-Razer_Razer_Kraken_Tournament_Edition_000000000000000000000000-event-if05 -> ../event10

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on Razer Synapse>

amaciuc, 2019-08-23:
Same problem to me.
The `0520` PID doesn't exist in supported headsets list for `openrazer`.

lamp216, 2019-08-30:
First of all thank u for ur work, i have the kraken tournament and tried to collect all the information u needed, if u need more infor just tell me

Please post the output of
    lsusb -d 1532: -v
```
Bus 001 Device 003: ID 1532:0520 Razer USA, Ltd 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0520 
  bcdDevice            0.88
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0198
    bNumInterfaces          6
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass         1 Audio
      bInterfaceSubClass      1 Control Device
      bInterfaceProtocol      0 
      iInterface              5 
      AudioControl Interface Descriptor:
        bLength                10
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdADC               1.00
        wTotalLength       0x0048
        bInCollection           2
        baInterfaceNr(0)        1
        baInterfaceNr(1)        2
      AudioControl Interface Descriptor:
        bLength                12
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID             1
        wTerminalType      0x0402 Headset
        bAssocTerminal          6
        bNrChannels             2
        wChannelConfig     0x0003
          Left Front (L)
          Right Front (R)
        iChannelNames           0 
        iTerminal               0 
      AudioControl Interface Descriptor:
        bLength                10
        bDescriptorType        36
        bDescriptorSubtype      6 (FEATURE_UNIT)
        bUnitID                 2
        bSourceID               1
[... 366 lines cut ...]
        bInterval               1
```
Please post the output of
ls -lah /dev/input/by-id/

```
total 0
drwxr-xr-x 2 root root 140 ago 30 23:26 .
drwxr-xr-x 4 root root 380 ago 30 23:26 ..
lrwxrwxrwx 1 root root  10 ago 30 23:02 usb-0c45_USB_Keyboard-event-if01 -> ../event11
lrwxrwxrwx 1 root root   9 ago 30 23:02 usb-0c45_USB_Keyboard-event-kbd -> ../event9
lrwxrwxrwx 1 root root  10 ago 30 23:02 usb-0c45_USB_Keyboard-if01-event-kbd -> ../event12
lrwxrwxrwx 1 root root   9 ago 30 23:02 usb-Logitech_USB_Optical_Mouse-event-mouse -> ../event8
lrwxrwxrwx 1 root root  10 ago 30 23:26 usb-Razer_Razer_Kraken_Tournament_Edition_000000000000000000000000-event-if05 -> ../event13
```
### Packet Captures
<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

[https://drive.google.com/drive/folders/1Dl7M41DKPmdi9ZBjD6_dQ2yma9RU5SGQ?usp=sharing](HERE)

### Synapse Screenshots
<Screenshots of all available options as seen on Razer Synapse>

[https://drive.google.com/drive/folders/15xiva5xqT9HbfMftseRN9-4EdbsLgylF?usp=sharing](HERE)

amaciuc, 2019-09-06:
Tell me, how did you install the driver and the app?

lamp216, 2019-09-15:
i installed both into a Windows VM

amaciuc, 2019-09-15:
I need it on Linux distribution (Debian)

lamp216, 2019-09-28:
oh i see, i had hope that THX could work somehow, thanks anyway for ur work dude

MatrixHulk, 2020-04-29:
Hello, will there be any support for the Razer Kraken 7.1 TE in the future?

z3ntu, 2020-05-01:
@MatrixHulk Also to you the question, what can we support with this device? It doesn't have any LEDs which is what we support on other headsets. Volume control etc. is purely software side and that should already work with standard tools.

jakogut, 2020-05-01:
@z3ntu The one thing I know of would be microphone feedback. I know this is a hardware feature, because it stays enabled after rebooting.

hubToxic, 2020-05-01:
Personally, I can't get my Razer Kraken TE to show up at all as a sound device on Debian 9. It can see the microphone, but not sound.

z3ntu, 2020-05-02:
@jakogut There are no normal "Razer protocol" packets sent to the device so I'm relatively sure this is still entirely software side.

@hubToxic The actual hardware support (apart from lighting and other misc functions) is out of scope for OpenRazer, this would have to be implemented in the Linux kernel itself. Maybe some other people can comment whether the device works out of the box (sound & microphone)?

cfeller, 2020-05-04:
> Maybe some other people can comment whether the device works out of the box (sound & microphone)?

When plugged into the USB DAC, the device "works" (sound & mic - I'm on Fedora 31) )although the gain on the speakers is relatively low. Volume output is as much as 50% lower than what it is in Windows. I have to turn the volume to 100% to even hear anything - it is fine, just not loud so it can be difficult to hear if there is a lot of background noise as these headphones aren't noise cancelling.

The buttons on the USB DAC do not work at all. The volume controls, nor the mute buttons have any function. I could bypass the DAC, but that is a hassle. I dual boot the machine between Windows and Fedora, and I don't want to have to constantly be rewiring along with my reboot. It would be nice to just have the USB DAC work.

FocusedWolf, 2020-05-08:
> Maybe some other people can comment whether the device works out of the box (sound & microphone)?

Tried on Pop!_OS 20.04 (installed), Manjaro 20.0 Gnome (live usb), and Fedora 32 (live usb).

No sound but mostly working buttons.

Pressing the mute button will toggle the red light circle on and off and the hiss from the microphone (sidetone) will noticeably stop when muted (the usb device obviously has a flash memory for storing settings because if you turn on Sidetone in Windows with Synapse 3 then you will have it in Linux).

Similarly if you turn on THX Spatial Audio in Windows the green LED will stay lit after reboot when booting into Linux.

Pressing the "+" and "-" a lot will eventually result in the beeping sound being played through the headphones indicating you are that the max/min limit.
THX Spatial Audio button does nothing in linux though (will not turn LEDs state... if it was on at the bios then it stays on in Linux).

----

Like i said before i get no sound through these in Linux. IDK how ppl get it to work but i'm guessing i need to toggle some Synapse 3 settings in Windows and reboot for them to work in Linux.

cfeller, 2020-05-13:
Updating/amending my previous comment (two up):

After FocusedWolf's comment (above) I did retest the headset with the DAC plugged in, and amazingly the buttons works now. When I first got the Kraken TE some time ago I was still running Fedora 29. I spent hours trying to get everything to work, and finally gave up. At that point I started following this issue.

I re-tested on Fedora 31 after FocusedWolf's comment and can confirm most things work. Additionally I upgraded to Fedora 32 and can confirm that it works here as well.

Sound up/down buttons work. Mute buttons work. If I move the game audio/chat audio slider on the side of the DAC to full "chat audio" the low speaker gain I mentioned above is mitigated. That is, it sounds as loud as it should (not sure if I'll have to readjust when I boot back into Windows). It would be nice to get the normal gain level without having to do this, but this is manageable. 

So really the only thing that doesn't appear to work at all is the THX spatial audio.

jaabberwocky, 2020-05-31:
Followed @FocusedWolf  and @cfeller  's advice above and the device works as expected on Ubuntu 20.04.

Even after installing the drivers and restarting, I still had issues getting it to work and did not have any sound. I removed the headset from the USB DAC and plugged it back in. After this, there was sound playing through the headset, and the microphone worked as expected. Sound up/down buttons work as usual, and the microphone mute button works too. Had to slide the DAC to "full chat audio" as was @cfeller 's experience.

demiverse, 2020-06-14:
@FocusedWolf thanks.
This headset works flawlesly, but no thx spatial sound. 
You don't even need to install openrazer, just slide your balance on "chat". If somehow you encouter problems, change output device to Digital Output (S/PDIF).

nicklaslof, 2021-07-11:
I did this to enable Sidetone (Microphone hardware feedback) on my Tournament edition in macOS Big Sur. It's in Rust but you get the idea of the USB-data needed:

```
fn enable_sidetone<T: UsbContext>(handle: &mut DeviceHandle<T>) -> Result<usize> {
    let timeout = Duration::from_secs(1);

    const REQUEST_TYPE: u8 = 0x21;
    const REQUEST: u8 = 0x09;
    const VALUE: u16 = 0x0204;
    const INDEX: u16 = 0x0005;
    const DATA: [u8; 37] = [
        0x04,0x40,0x01,0x12,0x77,0x0f,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00
    ]; // Data comes from Wireshark USB sniffing in Windows whilst enabling Sidetone in the Synapse control center application.
    handle.write_control(REQUEST_TYPE, REQUEST, VALUE, INDEX, &DATA, timeout)
}
```
full Rust-snippet: https://gitlab.com/-/snippets/2147708

jakogut, 2021-07-27:
@nicklaslof I did a little more sniffing and found that the sixth byte in that data sequence is the volume, from `0x00` to `0x10` scaled linearly.

Here's a Python snippet that does something similar:

```
import usb.core
import usb.util
import sys

def set_sidetone(vol=0x10):
    data = [0x04, 0x40, 0x01, 0x12,
            0x77, vol,  0x00, 0x00,
            0x00, 0x00, 0x00, 0x00,
            0x00, 0x00, 0x00, 0x00,
            0x00, 0x00, 0x00, 0x00,
            0x00, 0x00, 0x00, 0x00,
            0x00, 0x00, 0x00, 0x00,
            0x00, 0x00, 0x00, 0x00,
            0x00, 0x00, 0x00, 0x00]

    dev.ctrl_transfer(bmRequestType=0x21,
                      bRequest=0x09,
                      wValue=0x0204,
                      wIndex=0x0005,
                      data_or_wLength=data)

if __name__ == '__main__':
    dev = usb.core.find(idVendor=0x1532, idProduct=0x0520)

    if not dev:
        raise ValueError('Device not found')

    for config in dev:
        for i in range(config.bNumInterfaces):
            if dev.is_kernel_driver_active(i):
                dev.detach_kernel_driver(i)

    vol_pct = int(sys.argv[1])
    vol = round(vol_pct / 100 * 16) 
    set_sidetone(vol)

    for config in dev:
        for i in range(config.bNumInterfaces):
            dev.attach_kernel_driver(i)
```

Ryah, 2021-10-13:
Sorry if this isn't the right place, but I noticed people here were saying that the headset was working perfectly. On my distro (Pop OS), Pulseaudio only lets me enable either the Game output OR the Chat output, never both at the same time. 
[image]

Is there any way to enable both so I can have Discord going through the Chat output and everything else through the Game output and (hopefully) control them with the USB Controller?

xweskingx, 2022-08-03:
For the Kraken Tournament V2 i was able to enable sidetone with a slight tweak to @jakogut python script:
`   
    # 0x00 for enable 0x01 for disable 
    data = [0x00]

    dev.ctrl_transfer(bmRequestType=0x21,
            bRequest=0x01,
            wValue=0x0100,
            wIndex=0x0b00,
            data_or_wLength=data)

   # Set volume if enabling
    data = [0x00, vol]

    dev.ctrl_transfer(bmRequestType=0x21,
                      bRequest=0x01,
                      wValue=0x0200,
                      wIndex=0x0b00,
                      data_or_wLength=data)
`

Huskydog9988, 2024-05-27:
For those looking for an alternative solution to Ryah's method, I managed to hack around in pipewire to get a decent result. 

First run `pactl list cards`, and then find the device. Next run `pactl set-card-profile alsa_card.usb-Razer_Razer_USB_Audio_Controller_000000000000000000000000-00 pro-audio`, and this should roughly solve the issue, though it is hacky. (Both audio sources show up as Pro Audio, and Pro Audio 1 respectively.) You may also need to replace `usb-Razer_Razer_USB_Audio_Controller_000000000000000000000000-00` with whatever name is listed for you, I've seen this very quite a bit.

For the maintainers of openrazer: I hope someone can automate this process, and maybe have a much cleaner result then I currently do.

z3ntu, 2024-06-05:
@Huskydog9988 I'm not quite sure where this would fit but these pulseaudio/pipewire profiles are out of scope for OpenRazer fyi

bmbeverst, 2025-03-20:
Thanks for leaving this issue open!

Step-by-step guide for Razer Kraken Tournament Edition dual sound game and chat:

1. Check you are using PulseAudio, this probably doesn't work with pipewire:
    - `pactl info | grep "Server Name"` result should be `Server Name: pulseaudio`
2. Back up the `usb-gaming-headset.conf` configuration:
    - `sudo cp /usr/share/pulseaudio/alsa-mixer/profile-sets/usb-gaming-headset.conf{,.bak}`
    - Back is created at `/usr/share/pulseaudio/alsa-mixer/profile-sets/usb-gaming-headset.conf.bak`
3. Replace `usb-gaming-headset.conf` with the configuration in `steelseries-arctis-common-usb-audio.conf`. Basically, what @Ryah recommended in this [post](https://github.com/openrazer/openrazer/issues/836#issuecomment-941940032).
    - `sudo cp /usr/share/pulseaudio/alsa-mixer/profile-sets/steelseries-arctis-common-usb-audio.conf /usr/share/pulseaudio/alsa-mixer/profile-sets/usb-gaming-headset.conf`
4. Reboot
5. Enjoy your Game and Chat (Stereo) audio working on Linux!
    - Make sure to change the sound balance on the USB dongle when testing.
        - High beep is either all Game or Chat sound
        - Low beep is balanced between Game and Chat

If it didn't work or your sound is broken, then revert the changes with: `sudo cp /usr/share/pulseaudio/alsa-mixer/profile-sets/usb-gaming-headset.conf{.bak,}` (Notice the comma moved to the closing bracket) 

How it looks in KDE on Manjaro:
[image]

## Alternative udev rules change (Untested)
Or change the following files so they load the `steelseries-arctis-common-usb-audio.conf` for `ID 1532:0520 is for the Razer Kraken Tournament Edition`

- `/usr/lib/udev/rules.d/90-pipewire-alsa.rules`
- `/usr/lib/udev/rules.d/90-pulseaudio.rules`

The Razer Kraken line would need to be changed to `steelseries-arctis-common-usb-audio.conf` 

ATTRS{idVendor}=="1532", ATTRS{idProduct}=="0520", ENV{ACP_PROFILE_SET}="steelseries-arctis-common-usb-audio.conf"

[image]