openrazer/openrazer #840: Support for 3500 DPI OEM Elite Chroma DeathAdder Gaming USB Wired Mouse

Cobra-Ian, 2019-05-19:
### Device Info

* Name: deathadder 2013 [oem]
* Product Number: RZ01-00840100-R3C1
* Model Number: RZ01-0084 
### Device Information
```
Please post the output of
    lsusb -d 1532: -v
// this is an oem product so the vendor id is not 1532 but 1bcf
```
$ lsusb -d 1bcf: -v

Bus 001 Device 004: ID 1bcf:0005 Sunplus Innovation Technology Inc. Optical Mouse
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x1bcf Sunplus Innovation Technology Inc.
  idProduct          0x0005 Optical Mouse
  bcdDevice            0.90
  iManufacturer           0 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           34
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower               98mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      75
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0007  1x 7 bytes
        bInterval              10

```
Please post the output of
    ls -lah /dev/input/by-id/
```
$     ls -lah /dev/input/by-id/
total 0
drwxr-xr-x 2 root root 100 May 19 14:05 .
drwxr-xr-x 4 root root 380 May 19 14:05 ..
lrwxrwxrwx 1 root root   9 May 19 14:05 usb-1bcf_USB_Optical_Mouse-event-mouse -> ../event3
lrwxrwxrwx 1 root root   9 May 19 14:05 usb-1bcf_USB_Optical_Mouse-mouse -> ../mouse0
lrwxrwxrwx 1 root root   9 May 19 14:05 usb-CHICONY_HP_Basic_USB_Keyboard-event-kbd -> ../event2

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on Razer Synapse>

sorry i dont have access to a windows pc but i believe most of the relevant data can be found on the ebay product page especially considering its an oem version of an already supported device

https://www.ebay.co.uk/itm/3500-DPI-OEM-Elite-Chroma-DeathAdder-Gaming-USB-Wired-Mouse/264238902003?_trkparms=aid%3D111001%26algo%3DREC.SEED%26ao%3D1%26asc%3D20160908105057%26meid%3D896365c8a3da4ea482da2375d3641259%26pid%3D100675%26rk%3D1%26rkt%3D15%26mehot%3Dpp%26sd%3D264238902003%26itm%3D264238902003&_trksid=p2481888.c100675.m4236&_trkparms=pageci%3Ad80f3e0d-7a3a-11e9-8e21-74dbd1803149%7Cparentrq%3Ad04d1e9816a0a9cac9ff5e89ffbc8f7a%7Ciid%3A1

z3ntu, 2019-05-25:
This is not an original Razer device and likely not supported by Razer Synapse so there's not much we can do here.

closed 2019-05-25 by z3ntu

Cobra-Ian, 2019-05-25:
Case is not closed you have sold me a mouse which does not correctly work