openrazer/openrazer #842: Razer Basilisk Support?

0pLuS0, 2019-05-19:
### Distribution

N/A

### Support

- [N/A ] Have you tried rebooting?
- [ N/A] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Is there going to be any support soon for the Basilisk mouse?

Thanks

0pLuS0, 2019-05-19:
Sorry should of looked harder, saw this request for the mouse support too.

https://github.com/openrazer/openrazer/issues/433

I noticed mention of VirtualBox in that issue report, which I don't understand, what does VirtualBox have to do with getting support for this mouse in Linux? Unless I'm missing something with that issue... 

hmmm

lah7, 2019-05-20:
> what does VirtualBox have to do with getting support for this mouse in Linux?

Just to run a Windows virtual machine to capture USB packets from Razer Synapse, to understand the protocol of the mouse.

But... that device relies entirely on the software effects. [As commented here](https://github.com/openrazer/openrazer/issues/433#issuecomment-391497450), the OpenRazer daemon only supports hardware effects right now. There is a new userspace implementation in the works [(razer_test)](https://github.com/z3ntu/razer_test) that will eventually be the successor to the current daemon with proper support for software effects.

closed 2019-05-20 by lah7