openrazer/openrazer #843: Device not binding to driver although properly connected

AzimuthLI, 2019-05-20:
Hi,

I am using Razer Mamba wireless (Model NO. RC-30-027101) on Ubuntu 19.04.

I have installed openrazer-meta 2.5.0 and Polychromatic, but my device is not recognised by the daemon. 

Diagnose:

    lsusb | grep '1532:'
    Bus 001 Device 006: ID 1532:0073 Razer USA, Ltd 

And 

    sudo dkms status
    openrazer-driver, 2.5.0, 5.0.0-15-generic, x86_64: installed

And 
    
    sudo journalctl -b | grep -i razer_mount

gives nothing. 

I ran `modprobe razerkbd` and it succeeded. I also have put my user in plugdev group.

I also tried to mannuly bind the device following the instrunction in Wiki but it reports `no device` error in the last step. Then I got to know about the razer_mount script, and when I run the razer_mount script I got:

```
sudo sh razer_mount /sys/bus/hid/drivers/razerkbd 0003:1532:0073.0009
Driver /sys/bus/hid/drivers/razerkbd
Device_ID 0003:1532:0073.0009
Modprobing /sys/bus/hid/drivers/razerkbd
modprobe: FATAL: Module /sys/bus/hid/drivers/razerkbd not found in directory /lib/modules/5.0.0-15-generic
```

Anyidea what might go wrong? Thanks a lot in adv!

michaelschlies, 2019-05-23:
I don't believe the latest generation of Mamba Wireless / Mamba Elite are supported. I believe the 2018/2019 model has been reported in other mamba related issues as well

michaelschlies, 2019-05-23:
Oh, hey.. it was me that mentioned it! https://github.com/openrazer/openrazer/issues/505#issuecomment-476000048

AzimuthLI, 2019-05-23:
> Oh, hey.. it was me that mentioned it! [#505 (comment)](https://github.com/openrazer/openrazer/issues/505#issuecomment-476000048)

Ok good to know. Then I will just wait for the update.....

z3ntu, 2019-05-25:
1532:0073 is not supported by OpenRazer, so it can't work currently.

closed 2019-05-25 by z3ntu