openrazer/openrazer #844: Support for BlackWidow (2019)

asyd, 2019-05-20:
### Device Info

* Name: Razer BlackWidow
* Product Number: RZ03-02860100-R3M1
* Model Number: RZ03-0286
### Device Information
```
Bus 003 Device 002: ID 1532:0241 Razer USA, Ltd
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0241
  bcdDevice            2.00
  iManufacturer           1
  iProduct                2
  iSerial                 3
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0
      bInterfaceProtocol      1 Keyboard
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     179
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0016  1x 22 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0
      bInterfaceProtocol      2 Mouse
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1

```

```
total 0
drwxr-xr-x 2 root root 340 mai   20 08:31 .
drwxr-xr-x 4 root root 760 mai   20 08:31 ..
lrwxrwxrwx 1 root root  10 mai   20 08:31 usb-DAC_FOR_USB_FOCAL_XS-event-if02 -> ../event18
lrwxrwxrwx 1 root root  10 mai   20 08:31 usb-Logitech_G502_HERO_Gaming_Mouse_0E67385B3538-event-if01 -> ../event16
lrwxrwxrwx 1 root root  10 mai   20 08:31 usb-Logitech_G502_HERO_Gaming_Mouse_0E67385B3538-event-mouse -> ../event14
lrwxrwxrwx 1 root root  10 mai   20 08:31 usb-Logitech_G502_HERO_Gaming_Mouse_0E67385B3538-if01-event-kbd -> ../event15
lrwxrwxrwx 1 root root   9 mai   20 08:31 usb-Logitech_G502_HERO_Gaming_Mouse_0E67385B3538-mouse -> ../mouse2
lrwxrwxrwx 1 root root  10 mai   20 08:31 usb-Razer_Razer_BlackWidow-event-if01 -> ../event10
lrwxrwxrwx 1 root root   9 mai   20 08:31 usb-Razer_Razer_BlackWidow-event-kbd -> ../event8
lrwxrwxrwx 1 root root   9 mai   20 08:31 usb-Razer_Razer_BlackWidow-if01-event-kbd -> ../event9
lrwxrwxrwx 1 root root  10 mai   20 08:31 usb-Razer_Razer_BlackWidow-if02-event-mouse -> ../event13
lrwxrwxrwx 1 root root   9 mai   20 08:31 usb-Razer_Razer_BlackWidow-if02-mouse -> ../mouse1
lrwxrwxrwx 1 root root  10 mai   20 08:31 usb-Razer_Razer_Goliathus_Chroma-event-mouse -> ../event26
lrwxrwxrwx 1 root root   9 mai   20 08:31 usb-Razer_Razer_Goliathus_Chroma-if01-event-kbd -> ../event4
lrwxrwxrwx 1 root root   9 mai   20 08:31 usb-Razer_Razer_Goliathus_Chroma-if02-event-kbd -> ../event5
lrwxrwxrwx 1 root root   9 mai   20 08:31 usb-Razer_Razer_Goliathus_Chroma-mouse -> ../mouse0
lrwxrwxrwx 1 root root  10 mai   20 08:31 usb-Razer_Razer_Kraken_Tournament_Edition_000000000000000000000000-event-if05 -> ../event20

```

Thanks!

flarn2006, 2019-06-11:
How long do these usually take?

asyd, 2019-06-13:
According to 21dffa42d49f106bf55b8b4d45f564d6fa02c95f it doesn't look too complicated. 

However I have some doubts about some flags.

I'll give a try..

frikiluser, 2019-06-17:
I've patched some lines. It seems to work fine for me. Probably some adjustments are still needed.
I'ts my first Razer device, so I've never tested OpenRazer before.
 
https://github.com/frikiluser/openrazer

Hope it helps

frikiluser, 2019-06-17:
> However I have some doubts about some flags

I don't have a Windows system pcap.
May be I can help with some testing.

tildearrow, 2019-06-17:
@frikiluser You can open a pull request.

frikiluser, 2019-06-17:
Some final changes have been made.
Now i think all functionalities are enabled.

asyd, 2019-06-18:
@frikiluser 
I'm curious how you identified the closest model? By multiple try?

frikiluser, 2019-06-18:
> @frikiluser
> I'm curious how you identified the closest model? By multiple try?

Yes, I've tested some combinations until it start working. I think it worked at 3rd test. On successive tests I've added some missing functionalities (wave, breath and some more).

Not too hard, because the code is nice :-)

frikiluser, 2019-06-18:
@asyd btw, have you tested my commit?
I've vested it on Debian Buster with a brand new 1532:0241 keyboard. It has never been connected to a win/mac box.

asyd, 2019-06-19:
Let me try that (debian sid)

asyd, 2019-06-19:
Works perfectly! good job @frikiluser

asyd, 2019-06-19:
hmm sounds like I lost usage of fn + multimedia, I'll investigate

flarn2006, 2019-06-22:
Thanks for the good work. I'm having an issue where the left Alt key instead behaves as a third Control key—are you aware of that?

frikiluser, 2019-06-24:
> @flarn2006:
> Left Alt key instead behaves as a third Control key

Have you tested with 'xev'? I'll test tomorrow but I'm pretty sure Alt key it's working fine

flarn2006, 2019-06-24:
Yeah, I meant to say it actually did start working fine after I rebooted.
The first time, I just loaded it with modprobe.

PR #861 (2019-07-02): Add Razer BlackWidow 2019 support
Close: #844

It also fixes a recent typo in "install_files/udev/99-razer.rules". Missing leading 0.

I've tested all examples with successful results.
"advanced_effect.py" shows a big variety of effects, probably all available: breath_dual breath_random breath_single reactive spectrum static wave

Probably isn't perfect, because I haven't tested "official" driver.
closed 2019-07-02 by z3ntu