openrazer/openrazer #846: Error ever time i try to get this to run

joltox, 2019-05-23:
● openrazer-daemon.service - Daemon to manage razer devices in userspace
   Loaded: loaded (/usr/lib/systemd/user/openrazer-daemon.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Wed 2019-05-22 20:56:05 CDT; 25s ago
     Docs: man:openrazer-daemon(8)
  Process: 14328 ExecStart=/usr/bin/openrazer-daemon -Fv (code=exited, status=1/FAILURE)
 Main PID: 14328 (code=exited, status=1/FAILURE)

May 22 20:56:05 pop-os systemd[1604]: Starting Daemon to manage razer devices in userspace...
May 22 20:56:05 pop-os openrazer-daemon[14328]: Starting daemon.
May 22 20:56:05 pop-os openrazer-daemon[14328]: 2019-05-22 20:56:05 | razer                          | CRITICAL | User is not a member of the plugdev group
May 22 20:56:05 pop-os openrazer-daemon[14328]: Stopping daemon.
May 22 20:56:05 pop-os systemd[1604]: openrazer-daemon.service: Main process exited, code=exited, status=1/FAILURE
May 22 20:56:05 pop-os systemd[1604]: openrazer-daemon.service: Failed with result 'exit-code'.
May 22 20:56:05 pop-os systemd[1604]: Failed to start Daemon to manage razer devices in userspace.

joltox, 2019-05-23:
I am the only user on the computer, i have tried using my username and password but that does not resolve the issue. FYI I am quite new on Linux and first day on pop os

nightsky30, 2019-05-23:
Says:

`May 22 20:56:05 pop-os openrazer-daemon[14328]: 2019-05-22 20:56:05 | razer | CRITICAL | User is not a member of the plugdev group`

Try adding your user to the plugdev group and then rebooting:

`usermod -aG plugdev <username>`

joltox, 2019-05-23:
Tyvm! I took a break and came back to it, tried first with <> then w/o and
it worked, rebooted and it works!

closed 2019-05-23 by lah7