openrazer/openrazer #848: "Gaming Mode" disable alt-tab

bauer41s4, 2019-05-26:
### Distribution

### Support

- [ ] Have you tried rebooting?
- [ ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Hi when I enable gaming mode via polychromatic, my alt-tab functionality stops working. Oddly enough, it only stops working with the left alt key, right alt-tab still works. Any advice here? I really like the gaming mode feature, but don't want to lose left alt-tab functionality.

z3ntu, 2019-06-07:
I'm pretty sure this behavior is intended, but you can follow https://github.com/openrazer/openrazer/wiki/Using-the-keyboard-driver#key_alt_tab---alt-tab-key-combo-disable-toggle to enable alt-tab again.

bauer41s4, 2019-06-15:
When I look in the /sys/bus/hid/drivers/razerkbd directory, there are 3 different folders showing device IDs. Two of which contain the key_alt_tab file. Any idea as to which one I should be editing?

tildearrow, 2019-06-15:
Try the second one. The first one is used for non-game mode.

bauer41s4, 2019-06-16:
That worked -- thank you!

closed 2019-06-16 by bauer41s4