openrazer/openrazer #851: Razer abyssus 2014 isn't recognized

Teryss, 2019-06-02:
### Distribution

5.1.4-1-MANJARO

### Support

- [+] Have you tried rebooting?
- [+] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Hello!
The software (both polychromatic and razerCommander) doesn't detect my mouse (razer abyssus 2014), although I installed openrazer-meta (v. 2.5.0-1) from package manager. I went on troubleshooting page and anything I did didn't help. I've added user to the plugdev group (on command openrazer-daemon -Fv, it says Unable to lock on the pidfile), updated kernel and nothing helped.

I think that straight ahead I'll give logs and it'll be easier later.

sudo journalctl -b | grep -i razer_mount
`Jun 02 19:47:39 adam-pc razer_mount[450]: Driver razermouse
Jun 02 19:47:39 adam-pc razer_mount[451]: Device_ID 0003:1532:0042.0004
Jun 02 19:47:39 adam-pc razer_mount[452]: Modprobing razermouse
Jun 02 19:47:39 adam-pc systemd-udevd[344]: Process 'razer_mount razermouse 0003:1532:0042.0004' failed with exit code 1.
Jun 02 19:54:11 adam-pc razer_mount[2617]: Driver razermouse
Jun 02 19:54:11 adam-pc razer_mount[2618]: Device_ID 0003:1532:0042.0005
Jun 02 19:54:11 adam-pc razer_mount[2619]: Modprobing razermouse
Jun 02 19:54:11 adam-pc systemd-udevd[2605]: Process 'razer_mount razermouse 0003:1532:0042.0005' failed with exit code 1.`

sudo ~/.local/share/openrazer/logs/razer.log
`sudo: /home/adam/.local/share/openrazer/logs/razer.log: command not found`
           without sudo - acces denied

/lib/modules/$(uname -r)/build
`bash: /lib/modules/5.1.4-1-MANJARO/build: Is a directory`

If kernel module log needed, I'll post it later, because it's so long and makes my post not transparent.

z3ntu, 2019-06-02:
What happens when you run `sudo dkms install openrazer-driver/2.5.0` and `sudo modprobe razerkbd` ?

Teryss, 2019-06-02:
sudo dkms install openrazer-driver/2.5.0
`Creating symlink /var/lib/dkms/openrazer-driver/2.5.0/source ->
                 /usr/src/openrazer-driver-2.5.0

DKMS: add completed.
Error! echo
Your kernel headers for kernel 5.1.4-1-MANJARO cannot be found at
/usr/lib/modules/5.1.4-1-MANJARO/build or /usr/lib/modules/5.1.4-1-MANJARO/source.`

sudo modprobe razerkbd
`modprobe: FATAL: Module razerkbd not found in directory /lib/modules/5.1.4-1-MANJARO`

z3ntu, 2019-06-02:
Install your kernel headers please then :)

Teryss, 2019-06-02:
That worked thank you! 

For any manjaro users, u have to just install 

> linux49-headers

closed 2019-07-12 by z3ntu