openrazer/openrazer #852: Add method to retrieve current effect?

tildearrow, 2019-06-03:
The OpenRazer daemon provides no method to get a device's current effect. This causes minor problems for some applications:
- Polychromatic tells you "Try one..."
- RazerCommander shows no selected effect
- RazerGenie resets all devices to apparently "Breath Single"

The official Synapse software does remember the current effects and even sets the devices proper- I'll finish later

z3ntu, 2019-06-07:
Implemented in my razer_test project ;)

And contributions welcome!

tildearrow, 2019-06-07:
Here's my proposal:

A new per-device method called `getCurrentEffect` under `razer.device.lighting.chroma` (and other fields if needed), which returns 1 string (which is the effect name), 6 bytes (which are 2 colors), 1 byte (for speed) and 1 int32 (for direction).

CalcProgrammer1, 2019-06-16:
I would like this functionality as well.  I'm writing my own project to integrate a bunch of RGB drivers and devices together.  I'm not using the userspace part of OpenRazer but am using the kernel drivers.  I would like some way in the sysfs entries to get the current mode and the current colors.

tildearrow, 2019-06-17:
> I would like some way in the sysfs entries to get the current mode and the current colors.

While this may be possible, there are some problems to this approach:
- Where are we going to store the device's effect? By doing this in the daemon, we have razer.conf, but if done via sysfs we can't store the current effect... (which means having to reset the current effect to Spectrum every reboot and of course you don't want that)
- The devices don't have any commands to get the current effect (other than the DeathAdder Chroma/Kraken but they're weird ones), and even if they had, come on, we can't just expect the hardware to store everything for us (Synapse barely reads from the devices if I'm correct...).
  - Furthermore the reading mechanism is unreliable as sometimes it returns nothing (as an example, try reading the firmware version repeatedly, and at some point it'll return `v0.0`).
- If you're writing your own daemon, then why not write effect retrieval/storage support within it?

CalcProgrammer1, 2019-06-17:
I was assuming the Razer devices had a USB command to retrieve the information.  If they do not, then I agree it doesn't make much sense to add this to the sysfs entries as those should just implement the hardware capabilities.

tildearrow, 2019-10-25:
Still stale.
Closing. All activity moved to #888 anyways.

closed 2019-10-25 by tildearrow