openrazer/openrazer #853: No Mouse Colours On Ubuntu 18.04 (KDE NEON)

nshiell, 2019-06-04:
### Distribution
```uname -a```
Linux saturn 4.15.0-51-generic #55-Ubuntu SMP Wed May 15 14:27:21 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

```lsb_release -a```
No LSB modules are available.
Distributor ID: neon
Description:    KDE neon User Edition 5.15
Release:        18.04
Codename:       bionic

### Support

- [ X ] Have you tried rebooting?
- [ X ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

I am running KDE Neon (which is Ubuntu 18.04 with KDE Pasma desktop).
I have a Razer Mamba mouse, the colours worked OK before I upgraded to 18.04.
When I put the mouse on charge the green LEDs illuminate to indicate charge level.
When I go to Polychromatic, it sees the mouse and allows me to change colours.
However no colours illuminate on the mouse.

```tail -f ~/.local/share/openrazer/logs/razer.log```
Shows nothing unusual.

```sudo service razerd status```
Shows: 

```
...
Active: active (exited) since Tue 2019-06-04 18:29:18 BST; 16min ago
...
```

Any first used stable, i.e.:
sudo add-apt-repository ppa:openrazer/stable

then I tried moving to sudo add-apt-repository ppa:openrazer/daily
With reboots inbtween each attempt, any ideals?

z3ntu, 2019-07-12:
The razerd service is from https://github.com/mbuesch/razer , not from openrazer. Our service is called `openrazer-daemon.service`

Can you give me the output of the following commands? `sudo dkms install openrazer-driver/2.5.0` and `sudo modprobe razerkbd`

nshiell, 2019-08-10:
**sudo dkms install openrazer-driver/2.5.0**
`Module openrazer-driver/
2.5.0 already installed on kernel 4.15.0-55-generic/x86_64
`

**sudo modprobe razerkbd**
No output

Kirinkunhime, 2019-11-18:
Having a (very similar) issue in Ubuntu 18.10

> uname -a

Linux Kibum 4.18.0-25-lowlatency #26-Ubuntu SMP PREEMPT Mon Jun 24 10:17:48 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

> lsb_release -a

No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 19.04
Release:	19.04
Codename:	disco

**sudo modprobe razerkbd** does not deliver any output as well, when just trying to autocomplete it does not even show up - all other installed module (razercore, razerkraken, razermouse, razermousemat, razermug) do deliver the same 'empty' output as before.

nshiell, 2019-11-22:
I updated my desktop and went back to the GUI.
I noticed that the "Brightness" slider was set to the left most position i.e. brightness set to 0%
@Kirinkunhime please make sure all is up-to-date then check the brightness control

nshiell, 2019-11-22:
In fact I am going to close this issue as it's working for me now.

Some things we may need to consider, are brightness values ever set to zero by software updates (could happen) - this could cause issues related to this.

But as I can't recreate the issue now, it's not a bug for me anymore

closed 2019-11-22 by nshiell

Kirinkunhime, 2019-12-17:
As workaround I did enable the backlight again on windows (set to a non-zero value) and then it was editable again via openRazer - so works for me now as well.