openrazer/openrazer #854: BackWidow Chroma V2:  Custom lighting fails to illuminate space and fn keys

nightsky30, 2019-06-06:
### Distribution

Xubuntu 19.04
openrazer 2.5.0
RazerGenie 0.8.1

### Support

- [x] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

BackWidow Chroma V2:  Custom lighting fails to illuminate space and fn keys.  All other keys light up fine.  The same keys are NOT affected adversely with general lighting effects.  For reference to the hardware, there is 1 LED each under the space bar and fn key.

I did find [reference](https://github.com/openrazer/openrazer/blob/8f165e796fecf7e5a984346e8bf8034af52f0ec4/driver/razerchromacommon.c#L425) to a similar issue with the BlackWidow Ultimate 2016.  Have not been able to determine what is going on with these keys.

tildearrow, 2019-06-07:
Similar issue: https://github.com/z3ntu/RazerGenie/issues/35

nightsky30, 2019-06-07:
The very same checks out with the BlackWidow Chroma V2.  5_7, 5_12 as space and fn.

z3ntu, 2019-06-16:
Should be fixed with https://github.com/z3ntu/RazerGenie/pull/36 , but it's definitely not an openrazer issue.

closed 2019-06-16 by z3ntu