openrazer/openrazer #856: Support for Mamba Elite

tangyang111, 2019-06-08:
### Device Info

* Name: MAMBA ELITE
* Product Number: RZ01-02560100
* Model Number: RZ01-0256GAMING MOUSE

### Device Information
```
Please post the output of
    lsusb -d 1532: -v
```
Bus 003 Device 007: ID 1532:006c Razer USA, Ltd 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x006c 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1

```
Please post the output of
    ls -lah /dev/input/by-id/
```
total 0
drwxr-xr-x 2 root root 240 Jun  8 00:04 .
drwxr-xr-x 4 root root 740 Jun  8 00:04 ..
lrwxrwxrwx 1 root root  10 Jun  7 23:21 usb-Corsair_CORSAIR_K70_RGB_MK.2_SE_Mechanical_Gaming_Keyboard_0E01D01DAF3580E55B20042DF5001BC7-event-if00 -> ../event15
lrwxrwxrwx 1 root root  10 Jun  7 23:21 usb-Corsair_CORSAIR_K70_RGB_MK.2_SE_Mechanical_Gaming_Keyboard_0E01D01DAF3580E55B20042DF5001BC7-event-joystick -> ../event16
lrwxrwxrwx 1 root root  10 Jun  7 23:21 usb-Corsair_CORSAIR_K70_RGB_MK.2_SE_Mechanical_Gaming_Keyboard_0E01D01DAF3580E55B20042DF5001BC7-event-kbd -> ../event12
lrwxrwxrwx 1 root root   6 Jun  7 23:21 usb-Corsair_CORSAIR_K70_RGB_MK.2_SE_Mechanical_Gaming_Keyboard_0E01D01DAF3580E55B20042DF5001BC7-joystick -> ../js0
lrwxrwxrwx 1 root root  10 Jun  7 23:21 usb-Generic_Lenovo_EasyCamera_200901010001-event-if00 -> ../event20
lrwxrwxrwx 1 root root  10 Jun  8 00:04 usb-Razer_Razer_Mamba_Elite-event-if01 -> ../event10
lrwxrwxrwx 1 root root   9 Jun  8 00:04 usb-Razer_Razer_Mamba_Elite-event-mouse -> ../event6
lrwxrwxrwx 1 root root   9 Jun  8 00:04 usb-Razer_Razer_Mamba_Elite-if01-event-kbd -> ../event7
lrwxrwxrwx 1 root root  10 Jun  8 00:04 usb-Razer_Razer_Mamba_Elite-if02-event-kbd -> ../event11
lrwxrwxrwx 1 root root   9 Jun  8 00:04 usb-Razer_Razer_Mamba_Elite-mouse -> ../mouse1

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on Razer Synapse>
[image]

[image]

[image]

tangyang111, 2019-06-08:
[image]

nightsky30, 2019-06-09:
Please see #693

tangyang111, 2019-06-09:
Ok. Sorry for opening a duplicate issue.

closed 2019-06-09 by tangyang111