openrazer/openrazer #858: Poor documentation

Paradigm0001, 2019-06-10:
To quote the "Using the mouse driver" wiki page: "Inside the folder of the device binded to the driver there will be some of the following files"

Where is the folder? What does this mean? Its all excessively vague.

tildearrow, 2019-06-18:
The documentation at "Using the keyboard/mouse driver" is mostly for daemon developers, which is why it seems to be vague. Users generally use front-ends to control their devices. But I'll explain:

- The "folder" (actually a directory but they got used to the desktop metaphor) is one of the directories inside `/sys/bus/hid/drivers/razermouse` (e.g. `/sys/bus/hid/drivers/razermouse/0000:1532:0060.0001`). There are usually 3 of them per device, but only one contains the actual files that are used to control the mouse (usually the first one).
- "Binding to the driver" means assigning a device to the OpenRazer driver.

Paradigm0001, 2019-06-19:
@tildearrow Ah okay, thank you. I didnt mean to come across rude or belligerent, just wanted to understand what the wiki was referencing. Thank you for explaining this.