openrazer/openrazer #859: Support for Razer Blade Base Model 15 (2019) 1532:0246

noeRls, 2019-06-11:
### Device Info

* Name: Blade Base Model 15 (2019)
* Product Number: RZ09-03006*92
* Model Number: RZ09-0300

### Device Information
```
Please post the output of
    lsusb -d 1532: -v

Bus 001 Device 003: ID 1532:0246 Razer USA, Ltd 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0246 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1

```

```
Please post the output of
    ls -lah /dev/input/by-id/

drwxr-xr-x 2 root root 160 juin  11 22:23 .
drwxr-xr-x 4 root root 640 juin  11 22:23 ..
lrwxrwxrwx 1 root root  10 juin  11 22:23 usb-Azurewave_USB_Camera_0x0001-event-if00 -> ../event10
lrwxrwxrwx 1 root root   9 juin  11 22:23 usb-Razer_Razer_Blade-event-if01 -> ../event6
lrwxrwxrwx 1 root root   9 juin  11 22:23 usb-Razer_Razer_Blade-event-kbd -> ../event4
lrwxrwxrwx 1 root root   9 juin  11 22:23 usb-Razer_Razer_Blade-if01-event-kbd -> ../event5
lrwxrwxrwx 1 root root   9 juin  11 22:23 usb-Razer_Razer_Blade-if02-event-mouse -> ../event9
lrwxrwxrwx 1 root root   9 juin  11 22:23 usb-Razer_Razer_Blade-if02-mouse -> ../mouse0
```

### Packet Captures

### Synapse Screenshots

HalfAteMuffin, 2019-09-18:
Word. I'm running Parrot and am pretty bumed there is nothing for my model of laptop(ths model).

noeRls, 2019-09-18:
This is marked as "need pcap". I guess we need to provide a packet capture to make things going.

HalfAteMuffin, 2019-09-25:
I can get this, are there any instructions on what exactly to provide?

noeRls, 2019-09-25:
I don't exactly know how to do this, here's the link provided: https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol

HalfAteMuffin, 2019-09-27:
I'll check that out and see if I can get it done.

N-Argyle, 2019-11-01:
I can confirm that a pcap is necessary. I tried compiling drivers for other 2019 models for the 2019 base (0246), but they are in fact different. If I get time over the next few days I'll try and do the capture.

HalfAteMuffin, 2019-11-01:
Awesome please let me know if you need help with anything. Turns out, I’m
too new to linux to get the capture myself.

john-connort8001, 2019-12-14:
Openrazer does not work at all on 1532:0246 (2019 blade base model), you get device not recognized. I would care that much, but i Just want the damn keyboard light switch to stay at one colour, not asking for much :)

john-connort8001, 2019-12-21:
Can I help in any way to speed the process for this? Maybe use my razer blade for tests or anything?
Has anyone managed to control the keyboard lightning on with the current drivers?

ekenb, 2020-01-10:
> Can I help in any way to speed the process for this? Maybe use my razer blade for tests or anything?
> Has anyone managed to control the keyboard lightning on with the current drivers?

Same here, had I known the keyboard defaulted to chock demonstration like a cheap Christmas decoration I would have fixed it prior to removing windows.

If you have windows in a vm you can follow these instructions to get the devs what they need:
https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol

commit 94ae491b (2020-01-10): Add support for Razer Blade 15 (Mid 2019) Base

Fixes #859
closed 2020-01-10 by z3ntu