openrazer/openrazer #866: Support for Razer Laptop Stand Chroma

christianhujer, 2019-06-25:
### Device Info

* Name: Razer Laptop Stand Chroma
* Product Number: ?
* Model Number: RC21-0111

### Device Information
```
christian.hujer@Nelkinda-Blade-Stealth-2:~$ sudo lsusb -d 1532:0f0d -v

Bus 001 Device 017: ID 1532:0f0d Razer USA, Ltd 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0f0d 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Laptop Stand Chroma
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)
```

```
christian.hujer@Nelkinda-Blade-Stealth-2:~$ ls -lah /dev/input/by-id/
total 0
drwxr-xr-x 2 root root  820 Jun 25 16:56 .
drwxr-xr-x 4 root root 1.3K Jun 25 16:56 ..
lrwxrwxrwx 1 root root   10 Jun 25 11:24 usb-11918920-0079J9D8_USB_Camera-event-if00 -> ../event22
lrwxrwxrwx 1 root root   10 Jun 25 11:24 usb-ELAN_Touchscreen-event-if00 -> ../event26
lrwxrwxrwx 1 root root   11 Jun 25 11:24 usb-Intel_R__RealSense_TM__Camera_SR300_Intel_R__RealSense_TM__Camera_SR300_626206000901-event-if00 -> ../event262
lrwxrwxrwx 1 root root   11 Jun 25 11:24 usb-Intel_R__RealSense_TM__Camera_SR300_Intel_R__RealSense_TM__Camera_SR300_626206000901-event-if02 -> ../event265
lrwxrwxrwx 1 root root    9 Jun 25 11:24 usb-Microsoft_Microsoft®_Nano_Transceiver_v2.1-event-kbd -> ../event5
lrwxrwxrwx 1 root root    9 Jun 25 11:24 usb-Microsoft_Microsoft®_Nano_Transceiver_v2.1-if01-event-mouse -> ../event6
lrwxrwxrwx 1 root root    9 Jun 25 11:24 usb-Microsoft_Microsoft®_Nano_Transceiver_v2.1-if01-mouse -> ../mouse0
lrwxrwxrwx 1 root root    9 Jun 25 11:24 usb-Microsoft_Microsoft®_Nano_Transceiver_v2.1-if02-event-kbd -> ../event7
lrwxrwxrwx 1 root root   10 Jun 25 11:24 usb-Razer_Inc._Razer_Seiren_Pro_UC1712132400480-event-if03 -> ../event16
lrwxrwxrwx 1 root root   10 Jun 25 11:24 usb-Razer_Razer_Base_Station_Chroma-event-mouse -> ../event12
lrwxrwxrwx 1 root root   10 Jun 25 11:24 usb-Razer_Razer_Base_Station_Chroma-if01-event-kbd -> ../event13
lrwxrwxrwx 1 root root   10 Jun 25 11:24 usb-Razer_Razer_Base_Station_Chroma-if02-event-kbd -> ../event15
lrwxrwxrwx 1 root root    9 Jun 25 11:24 usb-Razer_Razer_Base_Station_Chroma-mouse -> ../mouse2
lrwxrwxrwx 1 root root   10 Jun 25 11:24 usb-Razer_Razer_Blade_Stealth-event-kbd -> ../event31
lrwxrwxrwx 1 root root   11 Jun 25 11:24 usb-Razer_Razer_Blade_Stealth-if01-event-kbd -> ../event256
lrwxrwxrwx 1 root root   10 Jun 25 11:24 usb-Razer_Razer_Blade_Stealth-if02-event-mouse -> ../event20
lrwxrwxrwx 1 root root    9 Jun 25 11:24 usb-Razer_Razer_Blade_Stealth-if02-mouse -> ../mouse1
lrwxrwxrwx 1 root root   11 Jun 25 11:24 usb-Razer_Razer_Firefly-event-mouse -> ../event263
lrwxrwxrwx 1 root root   11 Jun 25 11:24 usb-Razer_Razer_Firefly-if01-event-kbd -> ../event259
lrwxrwxrwx 1 root root   11 Jun 25 11:24 usb-Razer_Razer_Firefly-if02-event-kbd -> ../event260
lrwxrwxrwx 1 root root    9 Jun 25 11:24 usb-Razer_Razer_Firefly-mouse -> ../mouse4
lrwxrwxrwx 1 root root   10 Jun 25 11:24 usb-Razer_Razer_Goliathus_Chroma_Extended-event-mouse -> ../event28
lrwxrwxrwx 1 root root   11 Jun 25 11:24 usb-Razer_Razer_Goliathus_Chroma_Extended-if01-event-kbd -> ../event258
lrwxrwxrwx 1 root root   11 Jun 25 11:24 usb-Razer_Razer_Goliathus_Chroma_Extended-if02-event-kbd -> ../event257
lrwxrwxrwx 1 root root    9 Jun 25 11:24 usb-Razer_Razer_Goliathus_Chroma_Extended-mouse -> ../mouse3
lrwxrwxrwx 1 root root   10 Jun 25 16:48 usb-Razer_Razer_Lancehead-event-mouse -> ../event23
lrwxrwxrwx 1 root root   10 Jun 25 16:48 usb-Razer_Razer_Lancehead-if01-event-kbd -> ../event24
lrwxrwxrwx 1 root root   10 Jun 25 16:48 usb-Razer_Razer_Lancehead-if02-event-kbd -> ../event25
lrwxrwxrwx 1 root root    9 Jun 25 16:48 usb-Razer_Razer_Lancehead-mouse -> ../mouse5
lrwxrwxrwx 1 root root   11 Jun 25 16:56 usb-Razer_Razer_Laptop_Stand_Chroma-event-mouse -> ../event269
lrwxrwxrwx 1 root root   11 Jun 25 16:56 usb-Razer_Razer_Laptop_Stand_Chroma-if01-event-kbd -> ../event270
lrwxrwxrwx 1 root root   11 Jun 25 16:56 usb-Razer_Razer_Laptop_Stand_Chroma-if02-event-kbd -> ../event271
lrwxrwxrwx 1 root root    9 Jun 25 16:56 usb-Razer_Razer_Laptop_Stand_Chroma-mouse -> ../mouse9
lrwxrwxrwx 1 root root   10 Jun 25 11:24 usb-Razer_Razer_ManO_War-event-if03 -> ../event11
lrwxrwxrwx 1 root root   10 Jun 25 11:24 usb-Razer_Razer_Stargazer-Microphone_401649L00700344-event-if03 -> ../event14
lrwxrwxrwx 1 root root   11 Jun 25 16:56 usb-Razer_USB_Dongle_for_Razer_Lancehead-event-mouse -> ../event266
lrwxrwxrwx 1 root root   11 Jun 25 16:56 usb-Razer_USB_Dongle_for_Razer_Lancehead-if01-event-kbd -> ../event267
lrwxrwxrwx 1 root root   11 Jun 25 16:56 usb-Razer_USB_Dongle_for_Razer_Lancehead-if02-event-kbd -> ../event268
lrwxrwxrwx 1 root root    9 Jun 25 16:56 usb-Razer_USB_Dongle_for_Razer_Lancehead-mouse -> ../mouse8
```

### Packet Captures
Can't provide Razer Synapse Screenshots as I'm not using Windows at all ATM.
Will provide that later.

### Synapse Screenshots
Can't provide Razer Synapse Screenshots as I'm not using Windows at all ATM.
Will provide that later.

turbos-and-tacos, 2019-12-15:
Change the suffix to .pcap and wireshark should be able to show it.
[laptop_stand_static_G_to_R_to_G .txt](https://github.com/openrazer/openrazer/files/3965491/laptop_stand_static_G_to_R_to_G.txt)

z3ntu, 2019-12-16:
@turbos-and-tacos Thanks, pcap is good as well. But unfortunately the same thing as with the other pcap from yesterday.

VerstaelXV, 2021-03-26:
Hi, is there any new development in regards of support?

turbos-and-tacos, 2021-03-27:
I haven't messed with this in a few years now. I should take another stab at this.

VerstaelXV, 2021-03-27:
Would be awesome if you can get it to work 🤞🏻

VerstaelXV, 2021-04-01:
@turbos-and-tacos is there any news in the development, did you manage to get it to work ?

YannDev7, 2021-08-29:
Hey ! Is anyone still working on this feature please ? Also, do you need razer synapse screenshots ?
Thanks for creating this amazing software.

Daniel-Raymer, 2021-09-21:
Hey the laptop stand is still not detected, is anyone working on this?

z3ntu, 2021-10-01:
@turbos-and-tacos Do you still have the pcap lying around? GitHub shows a 404 error when accessing the link from above and I also don't find the file on my hard drive anymore. I could write some hints for the implementation from the pcap, but somebody with the actual device will have to get it actually working.

turbos-and-tacos, 2021-10-02:
@z3ntu I think I do. I should work on this, this weekend. I didn't have a lot of luck reverse engineering the packets the last time I took a stab at this. Some hints would be appreciated.

turbos-and-tacos, 2021-10-02:
[core_x_chroma.txt](https://github.com/openrazer/openrazer/files/7272064/core_x_chroma.txt)
Here is another pcap for the masses. Just change the file extension. (github doesn't allow pcaps or pcapng files)

islopma, 2021-11-05:
Looking forward to seeing support for this device. I have no idea on reverse engineering the device (awesome wiki BTW), but glad to help with anything else.

z3ntu, 2021-12-03:
Forgot to write, but the pcap doesn't say much except use transaction id 0x1f and `MATRIX_DIMS = [1, 16]`. The rest will be trial&error for someone with the device.

z3ntu, 2021-12-03:
@turbos-and-tacos Actually can you provide `lsusb | grep 1532` output from your machine and confirm you actually have a Laptop Stand Chroma? I have a suspicion that the pcap is from 1532:0f1a and not from 1532:0f0d that this issue is about.

psykher, 2022-02-25:
Do you know which device should be used here?

[image]

[image]

nhensh, 2022-05-25:
Would love to have this support, happy to spend some time helping as well 👍 

`system_profiler SPUSBDataType` gives me: 

Razer Laptop Stand Chroma:

 Product ID: 0x0f0d
 Vendor ID: 0x1532
 Version: 2.00
 Speed: Up to 12 Mb/s
 Manufacturer: Razer
 Location ID: 0x14214340 / 37
 Current Available (mA): 500
 Current Required (mA): 500
 Extra Operating Current (mA): 0

turbos-and-tacos, 2022-12-15:
Almost a year later now... but @z3ntu you are probably right. Has there been any work on the sw-based lighting front. I have an extended amount of free time and I'd like to do some real development on this.

`lsusb | grep 1532
Bus 007 Device 003: ID 1532:0f1a Razer USA, Ltd Core X Chroma
Bus 005 Device 009: ID 1532:0f0d Razer USA, Ltd Laptop Stand Chroma
Bus 005 Device 008: ID 1532:0c02 Razer USA, Ltd Goliathus Extended
Bus 005 Device 013: ID 1532:007e Razer USA, Ltd RC30-030502 Mouse Dock
Bus 005 Device 012: ID 1532:0088 Razer USA, Ltd Gaming Mouse [Basilisk Ultimate, Wireless, Receiver]
Bus 003 Device 002: ID 1532:0228 Razer USA, Ltd BlackWidow Elite
Bus 001 Device 006: ID 1532:026d Razer USA, Ltd Razer Blade
Bus 001 Device 009: ID 1532:0527 Razer USA, Ltd RZ04-0318 Gaming Headset [Kraken Ultimate]
`

StadSquared88, 2022-12-30:
First and foremost, this is an amazing piece of software! This also is the last device I cannot control. Is anyone actively working on this? Thank you so much for all the hard work.

closed 2023-01-15 by z3ntu