openrazer/openrazer #867: Support for Razer Abyssus Essential Chroma

jesusignazio, 2019-06-26:
### Device Info

* Name: Razer Abyssus Essential Chroma
* Product Number: RZ01-02160300
* Model Number: RZ01-0216

### Device Information
```
Please post the output of
    lsusb -d 1532: -v
```Bus 002 Device 002: ID 1532:006b Razer USA, Ltd 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x006b 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1

```
Please post the output of
    ls -lah /dev/input/by-id/
```
`drwxr-xr-x 2 root root 240 jun 26 23:47 ./
drwxr-xr-x 4 root root 620 jun 26 23:47 ../
lrwxrwxrwx 1 root root  10 jun 26 23:47 usb-Generic_USB2.0-CRW_20070818000000000-event-if02 -> ../event19
lrwxrwxrwx 1 root root   9 jun 26 23:47 usb-Razer_Razer_Abyssus_Essential-event-if01 -> ../event4
lrwxrwxrwx 1 root root   9 jun 26 23:47 usb-Razer_Razer_Abyssus_Essential-event-mouse -> ../event2
lrwxrwxrwx 1 root root   9 jun 26 23:47 usb-Razer_Razer_Abyssus_Essential-if01-event-kbd -> ../event3
lrwxrwxrwx 1 root root   9 jun 26 23:47 usb-Razer_Razer_Abyssus_Essential-if02-event-kbd -> ../event7
lrwxrwxrwx 1 root root   9 jun 26 23:47 usb-Razer_Razer_Abyssus_Essential-mouse -> ../mouse0
lrwxrwxrwx 1 root root   9 jun 26 23:47 usb-Razer_Razer_BlackWidow_Chroma-event-kbd -> ../event8
lrwxrwxrwx 1 root root   9 jun 26 23:47 usb-Razer_Razer_BlackWidow_Chroma-if01-event-kbd -> ../event9
lrwxrwxrwx 1 root root  10 jun 26 23:47 usb-Razer_Razer_BlackWidow_Chroma-if02-event-mouse -> ../event18
lrwxrwxrwx 1 root root   9 jun 26 23:47 usb-Razer_Razer_BlackWidow_Chroma-if02-mouse -> ../mouse1`

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on Razer Synapse>

titilambert, 2019-08-14:
It seems duplicate #612

z3ntu, 2019-08-18:
@titilambert Thanks. 

Duplicate of #612

closed 2019-08-18 by z3ntu