openrazer/openrazer #868: Cannot get Macros to work

888iee, 2019-06-29:
Distribution: Pop¹_OS 19.04 Linux Kernel 5.0.0

Bus 005 Device 003: ID 1532:011a Razer USA, Ltd BlackWidow Ultimate 2013

I've installed openrazer as stated in the description plus installed Polychromatic but I can't get the macros to work.
Polychromatic says I need to to use 

> FN + "Macro Key" -> then press they u want the macro to apply to -> enter your macro -> FN + "Macro Key"

In the originial razer handbook it's almost the same but you need to press Alt with FN + "Macro Key". This doesn't work either.
Macro LED is blinking after activation and stays on after entering the which function key to use.

In the end there is no macro executed or maybe not even exisiting.
On windows they M-Keys are working properly, in xev they look normal as well.
When entering `cat device_mode` it outputs `3:0`.

The steps I couldn't reproduce from the [setting up the keyboard driver](https://github.com/openrazer/openrazer/wiki/Setting-up-the-keyboard-driver) were the part of unbinding the driver 
`echo -n "0003:1532:011A.0006" > unbind`
> bash: unbind: No Permission

In this folder there is only a driver called 0003:046D:0A4D.0007
and binding the keyboard to razerkbd... gives the same error as above.
`echo -n "1" > mode_wave` and `cat device_type` and other are working properly.

I probably did something wrong but I don't know what as I'm new to Linux please help me understand.

tildearrow, 2019-06-29:
`mode_wave`? You probably are using a very old version of razer-drivers/OpenRazer.

888iee, 2019-07-01:
This line was given at the end of "Setting the keyboard drivers" section.

GetCurious, 2019-07-14:
i gave up getting my macros to work a year ago... looks like it'st still not gonna work huh...

kusayuzayushko, 2019-09-05:
>Macro LED is blinking after activation and stays on after entering the which function key to use

I have the same issue with LED stays on after entering which M key to assign the macro to. Using the 'openrazer-meta' package from AUR. Keyboard is "BlackWidow Ultimate 2013".

```
kusayu@desctop  ~  ls -l /sys/bus/hid/drivers
total 0
drwxr-xr-x 2 root root 0 Sep  5 13:25 hid-generic
drwxr-xr-x 2 root root 0 Sep  5 13:25 razerkbd
```
```
kusayu@desctop  ~  ls -l /sys/bus/hid/devices/
total 0
lrwxrwxrwx 1 root root 0 Sep  5 13:25 0003:09DA:9090.0001 -> ../../../devices/pci0000:00/0000:00:01.3/0000:03:00.0/usb1/1-3/1-3:1.0/0003:09DA:9090.0001
lrwxrwxrwx 1 root root 0 Sep  5 13:25 0003:09DA:9090.0002 -> ../../../devices/pci0000:00/0000:00:01.3/0000:03:00.0/usb1/1-3/1-3:1.1/0003:09DA:9090.0002
lrwxrwxrwx 1 root root 0 Sep  5 13:25 0003:1532:011A.0003 -> ../../../devices/pci0000:00/0000:00:07.1/0000:11:00.3/usb3/3-3/3-3:1.0/0003:1532:011A.0003
lrwxrwxrwx 1 root root 0 Sep  5 13:25 0003:1532:011A.0004 -> ../../../devices/pci0000:00/0000:00:07.1/0000:11:00.3/usb3/3-3/3-3:1.1/0003:1532:011A.0004
lrwxrwxrwx 1 root root 0 Sep  5 13:25 0003:1532:011A.0005 -> ../../../devices/pci0000:00/0000:00:07.1/0000:11:00.3/usb3/3-3/3-3:1.2/0003:1532:011A.0005
```
```
kusayu@desctop  ~  for i in {1..5}; do udevadm info -a /sys/bus/hid/devices/0003:1532:011A.000$i | grep bInterfaceNumber; done
Unknown device "/sys/bus/hid/devices/0003:1532:011A.0001": No such device
Unknown device "/sys/bus/hid/devices/0003:1532:011A.0002": No such device
    ATTRS{bInterfaceNumber}=="00"
    ATTRS{bInterfaceNumber}=="01"
    ATTRS{bInterfaceNumber}=="02"
```

So the correct device 0003:1532:011A.0005 is correctly binded to a razerkbd driver (just out of the box).

```
kusayu@desctop  ~  ls -l /sys/bus/hid/drivers/razerkbd
total 0
lrwxrwxrwx 1 root root    0 Sep  5 13:25 0003:1532:011A.0003 -> ../../../../devices/pci0000:00/0000:00:07.1/0000:11:00.3/usb3/3-3/3-3:1.0/0003:1532:011A.0003
lrwxrwxrwx 1 root root    0 Sep  5 13:25 0003:1532:011A.0004 -> ../../../../devices/pci0000:00/0000:00:07.1/0000:11:00.3/usb3/3-3/3-3:1.1/0003:1532:011A.0004
lrwxrwxrwx 1 root root    0 Sep  5 13:25 0003:1532:011A.0005 -> ../../../../devices/pci0000:00/0000:00:07.1/0000:11:00.3/usb3/3-3/3-3:1.2/0003:1532:011A.0005
--w------- 1 root root 4096 Sep  5 13:50 bind
lrwxrwxrwx 1 root root    0 Sep  5 13:50 module -> ../../../../module/razerkbd
--w------- 1 root root 4096 Sep  5 13:50 new_id
--w------- 1 root root 4096 Sep  5 13:25 uevent
--w------- 1 root root 4096 Sep  5 13:50 unbind
```

still pressing a macro key doesn't do anything. What am i missing?

Jachdich, 2019-09-05:
I'm having the same issue, good to see it's not just me. I'd really like to use macros, maybe I can hack-fix it into working with Python...

Jachdich, 2019-09-05:
Actually, it may be possible. Look at the this: https://polychromatic.app/docs/faqs/ and there is some python code you need to change for it to work. Haven't gotten it working myself but I hope to soon. @kusayuzayushko @GetCurious if you're interested.

Madoshakalaka, 2019-09-25:
Hi everybody,

This blog post can help. You can explicitly set macro keys with a short script:

https://madoshakalaka.github.io/2019/09/25/remap-keys-for-specific-device-on-ubuntu.html

closed 2020-02-23 by stale[bot]