openrazer/openrazer #876: Openrazer GPG error: public key is not available

kawmaiparis, 2019-07-04:
paris@paris-Blade:~$ sudo add-apt-repository ppa:openrazer/stable
 For users who would like a well-tested and confirmed working driver and daemon for supported devices.
 More info: https://launchpad.net/~openrazer/+archive/ubuntu/stable
Press [ENTER] to continue or Ctrl-c to cancel adding it.
 
Hit:1 http://th.archive.ubuntu.com/ubuntu bionic InRelease
Hit:2 http://th.archive.ubuntu.com/ubuntu bionic-updates InRelease                                      
Hit:3 http://th.archive.ubuntu.com/ubuntu bionic-backports InRelease                                    
Ign:4 http://dl.google.com/linux/chrome/deb stable InRelease                                   
Hit:5 http://dl.google.com/linux/chrome/deb stable Release                                     
Get:7 http://security.ubuntu.com/ubuntu bionic-security InRelease [88.7 kB]
Ign:8 http://download.opensuse.org/repositories/hardware:/razer/xUbuntu_18.04  InRelease        
Get:9 http://download.opensuse.org/repositories/hardware:/razer/xUbuntu_18.04  Release [988 B]
Get:10 http://download.opensuse.org/repositories/hardware:/razer/xUbuntu_18.04  Release.gpg [481 B]
Ign:10 http://download.opensuse.org/repositories/hardware:/razer/xUbuntu_18.04  Release.gpg
Hit:11 http://ppa.launchpad.net/openrazer/stable/ubuntu bionic InRelease
Hit:12 http://ppa.launchpad.net/polychromatic/stable/ubuntu bionic InRelease   
Hit:13 http://ppa.launchpad.net/teejee2008/ppa/ubuntu bionic InRelease         
Reading package lists... Done 
W: GPG error: http://download.opensuse.org/repositories/hardware:/razer/xUbuntu_18.04  Release: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 062F9FD4D6D11CE4
E: The repository 'http://download.opensuse.org/repositories/hardware:/razer/xUbuntu_18.04  Release' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

I'm getting this error when tried to install openrazer. Please help & Thank you!

z3ntu, 2019-07-04:
You didn't follow the instructions for (probably) adding the RazerGenie repository fully, see https://software.opensuse.org//download.html?project=hardware%3Arazer&package=razergenie (You also have to run the `You can add the repository key to apt. Keep in mind that the owner of the key may distribute updates, packages and repositories that your system will trust (more information). To add the key, run:` section)

z3ntu, 2019-07-20:
Closing because of inactivity.

closed 2019-07-20 by z3ntu