openrazer/openrazer #877: | CRITICAL | file is not owned by plugdev

BaronKrause, 2019-07-10:
So frequently I get the following in my openrazer log when I power on or reboot along with Polychromatic stating that there is an issue:

`2019-07-10 17:41:08 | razer                          | CRITICAL | Could not access /sys/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.0/0003:1532:0037.0001/device_type, file is not owned by plugdev`

It happens I would say almost 50% of the time at boot with the same Razer DeathAdder 2013 mouse on 2 different Pop OS systems.

I can always get it working again by simply unpluging, then repluging it in (and then right clicking on Polychromatic and selecting "restart daemon" to get that working).

Anyone know why this happens, and if there is a way to work around this issue to make it work at boot a bit more dependably?

Thank you!

z3ntu, 2019-07-11:
Weird, the udev rule triggers a script that chgrp's all files there... Can you give me the output of `ls -la /sys/bus/hid/drivers/razer*/0*/` ?

BaronKrause, 2019-07-11:
`'/sys/bus/hid/drivers/razermouse/0003:1532:0037.0003/':
total 0
drwxr-xr-x 5 root root    0 Jul 11 08:49 .
drwxr-xr-x 5 root root    0 Jul 11 08:48 ..
-r--r--r-- 1 root root 4096 Jul 11 08:49 country
-rw-rw---- 1 root root 4096 Jul 11 08:49 device_mode
-r--r----- 1 root root 4096 Jul 11 08:49 device_serial
-r--r----- 1 root root 4096 Jul 11 08:49 device_type
-rw-rw---- 1 root root 4096 Jul 11 08:49 dpi
lrwxrwxrwx 1 root root    0 Jul 11 08:48 driver -> ../../../../../../../../../../../../../bus/hid/drivers/razermouse
-r--r----- 1 root root 4096 Jul 11 08:49 firmware_version
drwxr-xr-x 3 root root    0 Jul 11 08:48 hidraw
drwxr-xr-x 3 root root    0 Jul 11 08:48 input
-rw-rw---- 1 root root 4096 Jul 11 08:49 logo_led_effect
-rw-rw---- 1 root root 4096 Jul 11 08:49 logo_led_rgb
-rw-rw---- 1 root root 4096 Jul 11 08:49 logo_led_state
-r--r--r-- 1 root root 4096 Jul 11 08:49 modalias
-rw-rw---- 1 root root 4096 Jul 11 08:49 poll_rate
drwxr-xr-x 2 root root    0 Jul 11 08:49 power
-r--r--r-- 1 root root 4096 Jul 11 08:49 report_descriptor
-rw-rw---- 1 root root 4096 Jul 11 08:49 scroll_led_effect
-rw-rw---- 1 root root 4096 Jul 11 08:49 scroll_led_rgb
-rw-rw---- 1 root root 4096 Jul 11 08:49 scroll_led_state
lrwxrwxrwx 1 root root    0 Jul 11 08:48 subsystem -> ../../../../../../../../../../../../../bus/hid
--w--w---- 1 root root 4096 Jul 11 08:49 test
-rw-r--r-- 1 root root 4096 Jul 11 08:48 uevent
-r--r----- 1 root root 4096 Jul 11 08:49 version

'/sys/bus/hid/drivers/razermouse/0003:1532:0037.0004/':
total 0
drwxr-xr-x 5 root root    0 Jul 11 08:49 .
drwxr-xr-x 5 root root    0 Jul 11 08:48 ..
-r--r--r-- 1 root root 4096 Jul 11 08:49 country
lrwxrwxrwx 1 root root    0 Jul 11 08:48 driver -> ../../../../../../../../../../../../../bus/hid/drivers/razermouse
drwxr-xr-x 3 root root    0 Jul 11 08:48 hidraw
drwxr-xr-x 3 root root    0 Jul 11 08:48 input
-r--r--r-- 1 root root 4096 Jul 11 08:49 modalias
drwxr-xr-x 2 root root    0 Jul 11 08:49 power
-r--r--r-- 1 root root 4096 Jul 11 08:49 report_descriptor
lrwxrwxrwx 1 root root    0 Jul 11 08:48 subsystem -> ../../../../../../../../../../../../../bus/hid
-rw-r--r-- 1 root root 4096 Jul 11 08:48 uevent

'/sys/bus/hid/drivers/razermouse/0003:1532:0037.0005/':
total 0
drwxr-xr-x 5 root plugdev    0 Jul 11 08:48 .
drwxr-xr-x 5 root root       0 Jul 11 08:48 ..
-r--r--r-- 1 root plugdev 4096 Jul 11 08:48 country
lrwxrwxrwx 1 root plugdev    0 Jul 11 08:48 driver -> ../../../../../../../../../../../../../bus/hid/drivers/razermouse
drwxr-xr-x 3 root plugdev    0 Jul 11 08:48 hidraw
drwxr-xr-x 3 root plugdev    0 Jul 11 08:48 input
-r--r--r-- 1 root plugdev 4096 Jul 11 08:48 modalias
drwxr-xr-x 2 root plugdev    0 Jul 11 08:48 power
-r--r--r-- 1 root plugdev 4096 Jul 11 08:48 report_descriptor
lrwxrwxrwx 1 root plugdev    0 Jul 11 08:48 subsystem -> ../../../../../../../../../../../../../bus/hid
-rw-r--r-- 1 root plugdev 4096 Jul 11 08:48 uevent
`

And after I unplug/replug to get it working (incase that matters):

`'/sys/bus/hid/drivers/razermouse/0003:1532:0037.0008/':
total 0
drwxr-xr-x 5 root plugdev    0 Jul 11 08:50 .
drwxr-xr-x 5 root root       0 Jul 11 08:50 ..
-r--r--r-- 1 root plugdev 4096 Jul 11 08:50 country
-rw-rw---- 1 root plugdev 4096 Jul 11 08:50 device_mode
-r--r----- 1 root plugdev 4096 Jul 11 08:50 device_serial
-r--r----- 1 root plugdev 4096 Jul 11 08:50 device_type
-rw-rw---- 1 root plugdev 4096 Jul 11 08:50 dpi
lrwxrwxrwx 1 root plugdev    0 Jul 11 08:50 driver -> ../../../../../../../../../../../../../bus/hid/drivers/razermouse
-r--r----- 1 root plugdev 4096 Jul 11 08:50 firmware_version
[... 20 lines cut ...]
drwxr-xr-x 5 root root       0 Jul 11 08:50 ..
-r--r--r-- 1 root plugdev 4096 Jul 11 08:50 country
lrwxrwxrwx 1 root plugdev    0 Jul 11 08:50 driver -> ../../../../../../../../../../../../../bus/hid/drivers/razermouse
drwxr-xr-x 3 root plugdev    0 Jul 11 08:50 hidraw
drwxr-xr-x 3 root plugdev    0 Jul 11 08:50 input
-r--r--r-- 1 root plugdev 4096 Jul 11 08:50 modalias
drwxr-xr-x 2 root plugdev    0 Jul 11 08:50 power
-r--r--r-- 1 root plugdev 4096 Jul 11 08:50 report_descriptor
lrwxrwxrwx 1 root plugdev    0 Jul 11 08:50 subsystem -> ../../../../../../../../../../../../../bus/hid
-rw-r--r-- 1 root plugdev 4096 Jul 11 08:50 uevent

'/sys/bus/hid/drivers/razermouse/0003:1532:0037.000A/':
total 0
drwxr-xr-x 5 root plugdev    0 Jul 11 08:50 .
drwxr-xr-x 5 root root       0 Jul 11 08:50 ..
-r--r--r-- 1 root plugdev 4096 Jul 11 08:50 country
lrwxrwxrwx 1 root plugdev    0 Jul 11 08:50 driver -> ../../../../../../../../../../../../../bus/hid/drivers/razermouse
drwxr-xr-x 3 root plugdev    0 Jul 11 08:50 hidraw
drwxr-xr-x 3 root plugdev    0 Jul 11 08:50 input
-r--r--r-- 1 root plugdev 4096 Jul 11 08:50 modalias
drwxr-xr-x 2 root plugdev    0 Jul 11 08:50 power
-r--r--r-- 1 root plugdev 4096 Jul 11 08:50 report_descriptor
lrwxrwxrwx 1 root plugdev    0 Jul 11 08:50 subsystem -> ../../../../../../../../../../../../../bus/hid
-rw-r--r-- 1 root plugdev 4096 Jul 11 08:50 uevent
`

z3ntu, 2019-07-11:
Weird... Is there anything significant in `sudo journalctl -b | grep -i razer_mount` ?

BaronKrause, 2019-07-11:
Fresh boot when it fails:

`Jul 11 17:45:21 acidburn razer_mount[1859]: Driver razermouse
Jul 11 17:45:21 acidburn razer_mount[1860]: Driver razermouse
Jul 11 17:45:21 acidburn razer_mount[1861]: Device_ID 0003:1532:0037.0003
Jul 11 17:45:21 acidburn razer_mount[1862]: Device_ID 0003:1532:0037.0004
Jul 11 17:45:21 acidburn razer_mount[1863]: Unbinding 0003:1532:0037.0004 from hid-generic
Jul 11 17:45:22 acidburn razer_mount[1985]: Binding 0003:1532:0037.0004 to razermouse
Jul 11 17:45:22 acidburn systemd-udevd[515]: Process 'razer_mount razermouse 0003:1532:0037.0004' failed with exit code 1.
Jul 11 17:45:22 acidburn razer_mount[1991]: Driver razermouse
Jul 11 17:45:22 acidburn razer_mount[1992]: Device_ID 0003:1532:0037.0005
Jul 11 17:45:22 acidburn razer_mount[1993]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0037.0005/
Jul 11 17:45:22 acidburn razer_mount[1995]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:0037.0005/
`

After Unplugging/Replugging:
`Jul 11 17:45:21 acidburn razer_mount[1859]: Driver razermouse
Jul 11 17:45:21 acidburn razer_mount[1860]: Driver razermouse
Jul 11 17:45:21 acidburn razer_mount[1861]: Device_ID 0003:1532:0037.0003
Jul 11 17:45:21 acidburn razer_mount[1862]: Device_ID 0003:1532:0037.0004
Jul 11 17:45:21 acidburn razer_mount[1863]: Unbinding 0003:1532:0037.0004 from hid-generic
Jul 11 17:45:22 acidburn razer_mount[1985]: Binding 0003:1532:0037.0004 to razermouse
Jul 11 17:45:22 acidburn systemd-udevd[515]: Process 'razer_mount razermouse 0003:1532:0037.0004' failed with exit code 1.
Jul 11 17:45:22 acidburn razer_mount[1991]: Driver razermouse
Jul 11 17:45:22 acidburn razer_mount[1992]: Device_ID 0003:1532:0037.0005
Jul 11 17:45:22 acidburn razer_mount[1993]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0037.0005/
Jul 11 17:45:22 acidburn razer_mount[1995]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:0037.0005/
Jul 11 17:47:54 acidburn razer_mount[5270]: Driver razermouse
Jul 11 17:47:54 acidburn razer_mount[5273]: Driver razermouse
Jul 11 17:47:54 acidburn razer_mount[5274]: Device_ID 0003:1532:0037.0008
Jul 11 17:47:54 acidburn razer_mount[5275]: Driver razermouse
Jul 11 17:47:54 acidburn razer_mount[5276]: Device_ID 0003:1532:0037.0009
Jul 11 17:47:54 acidburn razer_mount[5277]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0037.0008/
Jul 11 17:47:54 acidburn razer_mount[5278]: Device_ID 0003:1532:0037.000A
Jul 11 17:47:54 acidburn razer_mount[5280]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0037.0009/
Jul 11 17:47:54 acidburn razer_mount[5281]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0037.000A/
Jul 11 17:47:54 acidburn razer_mount[5283]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:0037.0008/
Jul 11 17:47:54 acidburn razer_mount[5286]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:0037.0009/
Jul 11 17:47:54 acidburn razer_mount[5288]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:0037.000A/
`

BaronKrause, 2019-07-11:
This is a fresh boot where it works like normal:
`Jul 11 17:55:31 acidburn razer_mount[839]: Driver razermouse
Jul 11 17:55:31 acidburn razer_mount[841]: Device_ID 0003:1532:0037.0003
Jul 11 17:55:31 acidburn razer_mount[843]: Unbinding 0003:1532:0037.0003 from hid-generic
Jul 11 17:55:31 acidburn razer_mount[900]: Binding 0003:1532:0037.0003 to razermouse
Jul 11 17:55:31 acidburn razer_mount[969]: Finished binding 0003:1532:0037.0003
Jul 11 17:55:31 acidburn razer_mount[973]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0037.0003/
Jul 11 17:55:31 acidburn razer_mount[980]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:0037.0003/
Jul 11 17:55:32 acidburn razer_mount[1005]: Driver razermouse
Jul 11 17:55:32 acidburn razer_mount[1004]: Driver razermouse
Jul 11 17:55:32 acidburn razer_mount[1006]: Device_ID 0003:1532:0037.0002
Jul 11 17:55:32 acidburn razer_mount[1007]: Device_ID 0003:1532:0037.0001
Jul 11 17:55:32 acidburn razer_mount[1008]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0037.0002/
Jul 11 17:55:32 acidburn razer_mount[1011]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:0037.0002/
Jul 11 17:55:32 acidburn razer_mount[1009]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0037.0001/
Jul 11 17:55:32 acidburn razer_mount[1016]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:0037.0001/
`

BaronKrause, 2019-07-11:
After some more testing it seems to happen SIGNIFICANTLY more when plugged in to the laptop through a thunderbolt 3 dock. When plugged in to a native USB 3 port it happens very infrequently (around the same rate as the second older PC I tested it on).

z3ntu, 2019-07-12:
This is the error:
```
Jul 11 17:45:22 acidburn razer_mount[1985]: Binding 0003:1532:0037.0004 to razermouse
Jul 11 17:45:22 acidburn systemd-udevd[515]: Process 'razer_mount razermouse 0003:1532:0037.0004' failed with exit code 1.
```
The code is just `	printf '%s' "$DEVICE_ID" > /sys/bus/hid/drivers/"$DRIVER"/bind` so apparently that fails for some reason... Was this always an issue or did this just start happening recently?

nightsky30, 2019-07-12:
I'm also experiencing this randomly when booting.  When it happens I have to unplug and plugin the device again to get around the issue.  It prevents my mouse and/or keyboard from responding:

Jul 12 09:50:23 w00t razer_mount[630]: Driver razermousemat
Jul 12 09:50:23 w00t razer_mount[629]: Driver razermouse
Jul 12 09:50:23 w00t razer_mount[644]: Device_ID 0003:1532:0060.0009
Jul 12 09:50:23 w00t razer_mount[645]: Device_ID 0003:1532:0C00.0002
Jul 12 09:50:23 w00t razer_mount[652]: Driver razerkbd
Jul 12 09:50:23 w00t razer_mount[653]: Device_ID 0003:1532:0221.0005
Jul 12 09:50:23 w00t razer_mount[654]: Unbinding 0003:1532:0221.0005 from hid-generic
Jul 12 09:50:23 w00t razer_mount[631]: Driver razermousemat
Jul 12 09:50:23 w00t razer_mount[647]: Unbinding 0003:1532:0060.0009 from hid-generic
Jul 12 09:50:23 w00t razer_mount[659]: Device_ID 0003:1532:0C00.0004
Jul 12 09:50:23 w00t razer_mount[668]: Unbinding 0003:1532:0C00.0004 from hid-generic
Jul 12 09:50:24 w00t razer_mount[1022]: Binding 0003:1532:0221.0005 to razerkbd
Jul 12 09:50:25 w00t razer_mount[1195]: Binding 0003:1532:0C00.0004 to razermousemat
Jul 12 09:50:25 w00t razer_mount[1247]: Finished binding 0003:1532:0221.0005
Jul 12 09:50:25 w00t razer_mount[1248]: Changing group /sys/bus/hid/drivers/razerkbd/0003:1532:0221.0005/
Jul 12 09:50:25 w00t razer_mount[1250]: Changed group /sys/bus/hid/drivers/razerkbd/0003:1532:0221.0005/
Jul 12 09:50:25 w00t razer_mount[1346]: Finished binding 0003:1532:0C00.0004
Jul 12 09:50:25 w00t razer_mount[1347]: Changing group /sys/bus/hid/drivers/razermousemat/0003:1532:0C00.0004/
Jul 12 09:50:25 w00t razer_mount[1349]: Changed group /sys/bus/hid/drivers/razermousemat/0003:1532:0C00.0004/
Jul 12 09:50:25 w00t razer_mount[1350]: Binding 0003:1532:0060.0009 to razermouse
**Jul 12 09:50:25 w00t systemd-udevd[488]: Process 'razer_mount razermouse 0003:1532:0060.0009' failed with exit code 1.**
Jul 12 09:50:26 w00t razer_mount[1358]: Driver razermousemat
Jul 12 09:50:26 w00t razer_mount[1359]: Device_ID 0003:1532:0C00.0003
Jul 12 09:50:26 w00t razer_mount[1360]: Changing group /sys/bus/hid/drivers/razermousemat/0003:1532:0C00.0003/
Jul 12 09:50:26 w00t razer_mount[1362]: Changed group /sys/bus/hid/drivers/razermousemat/0003:1532:0C00.0003/
Jul 12 09:50:26 w00t razer_mount[1370]: Driver razermouse
Jul 12 09:50:26 w00t razer_mount[1371]: Driver razermouse
Jul 12 09:50:26 w00t razer_mount[1372]: Driver razerkbd
Jul 12 09:50:26 w00t razer_mount[1373]: Driver razerkbd
Jul 12 09:50:26 w00t razer_mount[1374]: Device_ID 0003:1532:0060.0008
Jul 12 09:50:26 w00t razer_mount[1375]: Device_ID 0003:1532:0060.000A
Jul 12 09:50:26 w00t razer_mount[1376]: Device_ID 0003:1532:0221.0007
Jul 12 09:50:26 w00t razer_mount[1379]: Changing group /sys/bus/hid/drivers/razerkbd/0003:1532:0221.0007/
Jul 12 09:50:26 w00t razer_mount[1378]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0060.000A/
Jul 12 09:50:26 w00t razer_mount[1377]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0060.0008/
Jul 12 09:50:26 w00t razer_mount[1380]: Device_ID 0003:1532:0221.0006
Jul 12 09:50:26 w00t razer_mount[1384]: Changing group /sys/bus/hid/drivers/razerkbd/0003:1532:0221.0006/
Jul 12 09:50:26 w00t razer_mount[1385]: Changed group /sys/bus/hid/drivers/razerkbd/0003:1532:0221.0007/
Jul 12 09:50:26 w00t razer_mount[1386]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:0060.000A/
Jul 12 09:50:26 w00t razer_mount[1387]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:0060.0008/
Jul 12 09:50:26 w00t razer_mount[1389]: Changed group /sys/bus/hid/drivers/razerkbd/0003:1532:0221.0006/
Jul 12 09:50:53 w00t razer_mount[1699]: Driver razerkbd
Jul 12 09:50:53 w00t razer_mount[1700]: Driver razerkbd
Jul 12 09:50:53 w00t razer_mount[1701]: Driver razerkbd
Jul 12 09:50:53 w00t razer_mount[1702]: Device_ID 0003:1532:0221.000D
Jul 12 09:50:53 w00t razer_mount[1703]: Device_ID 0003:1532:0221.000B
Jul 12 09:50:53 w00t razer_mount[1704]: Device_ID 0003:1532:0221.000C
Jul 12 09:50:53 w00t razer_mount[1705]: Changing group /sys/bus/hid/drivers/razerkbd/0003:1532:0221.000D/
Jul 12 09:50:53 w00t razer_mount[1706]: Changing group /sys/bus/hid/drivers/razerkbd/0003:1532:0221.000B/
Jul 12 09:50:53 w00t razer_mount[1707]: Changing group /sys/bus/hid/drivers/razerkbd/0003:1532:0221.000C/
Jul 12 09:50:53 w00t razer_mount[1711]: Changed group /sys/bus/hid/drivers/razerkbd/0003:1532:0221.000B/
Jul 12 09:50:53 w00t razer_mount[1712]: Changed group /sys/bus/hid/drivers/razerkbd/0003:1532:0221.000D/
Jul 12 09:50:53 w00t razer_mount[1713]: Changed group /sys/bus/hid/drivers/razerkbd/0003:1532:0221.000C/
Jul 12 09:50:56 w00t razer_mount[1755]: Driver razermouse
Jul 12 09:50:56 w00t razer_mount[1758]: Driver razermouse
Jul 12 09:50:56 w00t razer_mount[1759]: Driver razermouse
Jul 12 09:50:56 w00t razer_mount[1760]: Device_ID 0003:1532:0060.000F
Jul 12 09:50:56 w00t razer_mount[1761]: Device_ID 0003:1532:0060.0010
Jul 12 09:50:56 w00t razer_mount[1762]: Device_ID 0003:1532:0060.000E
Jul 12 09:50:56 w00t razer_mount[1763]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0060.000F/
Jul 12 09:50:56 w00t razer_mount[1764]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0060.0010/
Jul 12 09:50:56 w00t razer_mount[1765]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0060.000E/
Jul 12 09:50:56 w00t razer_mount[1769]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:0060.000F/
[... cut ...]
Jul 12 09:51:04 w00t razer_mount[1822]: Changed group /sys/bus/hid/drivers/razermousemat/0003:1532:0C00.0012/
Jul 12 09:51:04 w00t razer_mount[1821]: Changed group /sys/bus/hid/drivers/razermousemat/0003:1532:0C00.0013/
Jul 12 09:51:04 w00t razer_mount[1823]: Changed group /sys/bus/hid/drivers/razermousemat/0003:1532:0C00.0011/

Not sure what @BaronKrause has experienced, but in my instance it seems it started after I bought my firefly.  It hadn't happened before then.  Not sure if it is my firefly, the fact that I have 3 Razer devices now instead of 2, or something else coincidentally.

BaronKrause, 2019-07-13:
OK with extended testing on my second system, and also on the first, but plugged directly in without going through the thunderbolt dock, it seems it works almost every time as it should.

There is a roughly 8-10% rate that Polychromatic starts up with the red icon still, but it can be fixed by Just right-clicking it and telling it to retry. Something that doesn't work with the previously listed issue. I assumed the issues were the same and that it was just a bit worse on the dock, but this seems like it is more of a Polychromatic starting before openrazer finishes binding sorta thing.

So, it only happens when on the dock. Thunderbolt devices aren't connected all the time like a USB3 dock, it is initiated and authorized by the boltd thunderbolt daemon, so devices connected through it will be connected to the system slightly delayed, (not perceptibly though, since you can move the mouse and type as soon as the login window appears).

Do you think the 50% binding issue when using the dock is caused by the slight delay, or something else?

Can this be worked around aside from just using the side ports? Its not a big deal at all, just docks kind of lose their purpose if you still need to  plug other things into the system manually.

z3ntu, 2019-07-13:
Thanks for testing but I have no idea what could be done to fix this... You can try putting a sleep statement at the top of `/usr/lib/udev/razer_mount` (or without the `/usr` on Debian+derivatives) and see if that helps...

Jan200101, 2019-09-06:
I had the exact same issue, I don't know what exactly caused this but removing /var/lib/dkms/openrazer-driver/ and reinstalling openrazer fixed it

vincenttaglia, 2019-09-15:
Been having this issue a lot as I've been rebuilding deb packs and reinstalling. @Jan200101's suggestion made the frequency a whole lot less, but it still happened every once in a while.

andrewschott, 2019-10-29:
I am having the same issue.  Replugging the devices in seems to fix things.  Is there a way to automate this at boot as it used to work?  BTW, all the devices (keyboard, gaming keypad, mouse, mat) are directly plugged into the USB3 ports on my motherboard of a tower system.  Reinstallation of the OpenSUSE TW rpms does not work.  I tried removal, reboot, install, reboot -- no dice.

ruvaldak, 2019-11-24:
This also happens to me. No dock or anything, using a desktop PC myself. Running Manjaro, on KDE Plasma

hawajka76, 2020-04-11:
I just encountered the same issue on Ubuntu 18.04 with the Deathadder Elite (RZ01-02010100), with the latest daily openrazer build and polychromatic. Replugging the mouse fixed the issue at least temporarily.

psidhu, 2020-07-08:
I can reproduce this 100% of the time by doing the following:
1. `sudo rmmod razerkbd`
2. `kill -9 $(pidof openrazer-daemon)`
3. `sudo insmod ./driver/razerkbd.ko` # This is my built version, from mainline
4. `./daemon/run_openrazer_daemon.py`
5. See issue. Fix by re-plugging keyboard (huntsman TE in my case).

Might help whoever might work on this. Without having dug into this _at all_, I suspect it could be the driver remove function.

z3ntu, 2020-07-08:
@psidhu Yes, this is kind of expected, the udev rules only get triggered on device add/remove and not on a kernel module add/remove so I think that this issue is about something else. But if you can fix this behavior it would be appreciated.