openrazer/openrazer #878: Device not recognised (razer taipan)

kik4444, 2019-07-12:
### Distribution

4.19.56-1-MANJARO

### Support

- [ ] Have you tried rebooting? - YES
- [ ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page? - YES

I'm running Manjaro and my razer device is the taipan mouse. I installed openrazer-meta and polychromatic and restarted. I've added myself to the plugdev group but it still says my device is not recognised. 

Here is the razer.log - https://pastebin.com/WFSMPXyT

The result of `sudo journalctl -b | grep -i razer_mount` is:

`
юли 12 14:30:33 Kiko-LM razer_mount[509]: Driver razermouse
юли 12 14:30:33 Kiko-LM razer_mount[508]: Driver razermouse
юли 12 14:30:33 Kiko-LM razer_mount[512]: Device_ID 0003:1532:0034.0006
юли 12 14:30:33 Kiko-LM razer_mount[511]: Device_ID 0003:1532:0034.0005
юли 12 14:30:33 Kiko-LM razer_mount[514]: Modprobing razermouse
юли 12 14:30:33 Kiko-LM razer_mount[515]: Modprobing razermouse
юли 12 14:30:33 Kiko-LM systemd-udevd[436]: Process 'razer_mount razermouse 0003:1532:0034.0006' failed with exit code 1.
юли 12 14:30:33 Kiko-LM razer_mount[520]: Driver razermouse
юли 12 14:30:33 Kiko-LM systemd-udevd[407]: Process 'razer_mount razermouse 0003:1532:0034.0005' failed with exit code 1.
юли 12 14:30:33 Kiko-LM razer_mount[522]: Device_ID 0003:1532:0034.0004
юли 12 14:30:33 Kiko-LM razer_mount[526]: Modprobing razermouse
юли 12 14:30:33 Kiko-LM systemd-udevd[430]: Process 'razer_mount razermouse 0003:1532:0034.0004' failed with exit code 1.
`

And `sudo dkms install openrazer-driver/1.0.0` returns:
`Error! Could not find module source directory.
Directory: /usr/src/openrazer-driver-1.0.0 does not exist.`

z3ntu, 2019-07-12:
What's the output from the command `sudo dkms install openrazer-driver/2.5.0` ? Also please try running `sudo modprobe razerkbd`

kik4444, 2019-07-12:
The first command actually returned

`Creating symlink /var/lib/dkms/openrazer-driver/2.5.0/source ->
                 /usr/src/openrazer-driver-2.5.0

DKMS: add completed.
Error! echo
Your kernel headers for kernel 4.19.56-1-MANJARO cannot be found at
/usr/lib/modules/4.19.56-1-MANJARO/build or /usr/lib/modules/4.19.56-1-MANJARO/source.`

And `sudo modprobe razerkbd` returned:

`modprobe: FATAL: Module razerkbd not found in directory /lib/modules/4.19.56-1-MANJARO`

z3ntu, 2019-07-12:
Please install your kernel headers as written at https://wiki.archlinux.org/index.php/Dynamic_Kernel_Module_Support#Installation .

kik4444, 2019-07-12:
How should I do it for my case? I don't know what to look for, I'm not familiar with how kernel headers work.

z3ntu, 2019-07-12:
Depends on what kernel package you have installed and are using. Use that and append `-headers` to the name. It's probably `linux419-headers` or something like that - Manjaro has weird package names for that.

kik4444, 2019-07-12:
Ah yes it was exactly linux419-headers. Now it works, but it's a bit underwhelming. I don't see where I can remap the extra keys.

z3ntu, 2019-07-12:
Key remapping isn't supported yet. Closing as the setup issue is solved.

closed 2019-07-12 by z3ntu