openrazer/openrazer #879: [Manjaro 18.0.4 - Cynosa Chroma] Device Not Recognized

cyclowns, 2019-07-14:
### Distribution
Manjaro Linux 18.0.4 Illyria
Kernel is 4.1.56-1-MANJARO

### Support

- [x] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Using openrazer with polychromatic. Says 'device not recognized' despite the device appearing in lsusb (Razer Cynosa Chroma, 1532:022a), openrazer-daemon running just fine (installed from the AUR using trizen), adding myself to the plugdev group, and rebooting several times.

I'm unable to disable secure boot as my BIOS doesnt even allow me to (dell A05)

## logs
```bash
$ lsusb | grep "1532:"
Bus 002 Device 005: ID 1532:022a Razer USA, Ltd
```

```bash
$ sudo dkms install openrazer-driver/2.5.0
Creating symlink /var/lib/dkms/openrazer-driver/2.5.0/source ->
                 /usr/src/openrazer-driver-2.5.0

DKMS: add completed.
Error! echo
Your kernel headers for kernel 4.19.56-1-MANJARO cannot be found at
/usr/lib/modules/4.19.56-1-MANJARO/build or /usr/lib/modules/4.19.56-1-MANJARO/source.

(this seems like an issue but I dont know how to resolve that myself)
```

```openrazer-daemon``` log: https://pastebin.com/d3MBRhuS

z3ntu, 2019-07-14:
Install the package linux419-headers which are your kernel headers, which are missing, as indicated with the `Your kernel headers for kernel 4.19.56-1-MANJARO cannot be found at ...` message :)

cyclowns, 2019-07-14:
Installing linux419-headers and rebooting worked!

closed 2019-07-14 by cyclowns