openrazer/openrazer #883: Support for razer Death Adder

SheenBR, 2019-07-25:
**Device Info**

- Name: Razer Death Adder

- Product Number: RZ01-00850100-R3C1

- Model Number: RZ01-0085

**Device Info**
`Bus 004 Device 002: ID 1532:0038 Razer USA, Ltd 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0038 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer DeathAdder 1800
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      0 None
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               8
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      41
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               8
Device Status:     0x0000
  (Bus Powered)
`
`leandro@Casa:~$ ls -lah /dev/input/by-id/
total 0
drwxr-xr-x 2 root root 120 jul 25 08:11 .
drwxr-xr-x 4 root root 520 jul 25 08:11 ..
lrwxrwxrwx 1 root root   9 jul 25 08:11 usb-Razer_Razer_DeathAdder_1800-event-mouse -> ../event3
lrwxrwxrwx 1 root root   9 jul 25 08:11 usb-Razer_Razer_DeathAdder_1800-if01-event-kbd -> ../event4
lrwxrwxrwx 1 root root   9 jul 25 08:11 usb-Razer_Razer_DeathAdder_1800-if02-event-kbd -> ../event5
lrwxrwxrwx 1 root root   9 jul 25 08:11 usb-Razer_Razer_DeathAdder_1800-mouse -> ../mouse0
`

**Packet Captures**
I'm not using Windows at all at the moment

**Synapse Screenshots**
I'm not using Windows at all at the moment

jiazhoulvke, 2019-09-04:
**Packet Captures**

[Razer_DeathAdder_1800_Crossfire_2014_Edition.pcap.zip](https://github.com/openrazer/openrazer/files/3574429/Razer_DeathAdder_1800_Crossfire_2014_Edition.pcap.zip)

**Synapse Screenshots**

[image]
[image]
[image]
[image]

nmasnadithya, 2019-09-15:
I also have this device. Would love to help with testing and getting more info. Also, I might be able to help with the development if someone is willing to give me some guidance.

z3ntu, 2019-09-24:
@nmasnadithya The first step is adding support to the kernel driver.

For the general outline what's needed, you can look at a commit like d56b8f1bfa9b89254f3c49c41a52d8b39285036d (ignore the changes in `daemon/` in the beginning). And do your edits in `driver/razermouse_driver.c` (instead of the `razerkbd_driver.c`). In the driver file it's mostly adding case statements to the correct sections, which in this case are:

DPI: https://github.com/openrazer/openrazer/blob/master/driver/razermouse_driver.c#L898 and https://github.com/openrazer/openrazer/blob/master/driver/razermouse_driver.c#L1013 (the DeathAdder 1800 also uses just a byte for DPI)
Poll rate: normal as with every device (no need to add a case there)
LED effects: the device only has a "set led state" with on and off for the logo LED, so you can use `logo_led_state`

So for the `device_remove_files` and `CREATE_DEVICE_FILE` sections you can use the same as e.g. the Abyssus 1800 so just add the case statement for your device there.

If you have any further question, you can join our community chat, see https://github.com/openrazer/openrazer/wiki/Community-chat for details, as it will be quicker there to get an answer from me :)

nmasnadithya, 2019-10-23:
I'm almost done with this. There are some minor issues that needs to be sorted. Planing to submit the PR tonight.
Check this branch if someone wants to test my changes before they get merged.
[https://github.com/nmasnadithya/openrazer/tree/feature_deathadder1800](https://github.com/nmasnadithya/openrazer/tree/feature_deathadder1800)

commit 6d792635 (2020-01-05): Add Razer DeathAdder 1800 support

Fixes #883
PR #939 (2020-01-05): Adding support for Razer DeathAdder 1800
Fixes #883
closed 2020-01-05 by z3ntu