openrazer/openrazer #884: openrazer-daemon is not working with kernel 5.3

Stefan538871, 2019-07-26:
I get the following error in razergenie menu:
Linux detected connected devices but the daemon didn't. This could be either due to a permission problem or a kernel module problem.

Since 5.3 isnt released yet is is probably safe to assume that they changed something which broke something with openrazer-daemon. I checked and the daemon was running, also tried to restart the service with --as-root.
I use opensuse TW and used this guide to install newer kernels https://www.addictivetips.com/ubuntu-linux-tips/install-new-linux-kernel-versions-on-opensuse-leap/

Cant check the exact 5.3 version, because i probably just broke my system while trying to downgrade the kernel. Saving data and reinstall it is

parkerlreed, 2019-07-27:
What does the daemon log have to say? That razergenie error isn't descriptive.

z3ntu, 2019-07-27:
You're likely missing the kernel headers package for the 5.3 kernel that you installed.

Stefan538871, 2019-07-28:
Unfortunatly i cant access the log files anymore. Since i tried to downgrade to 5.2 and there went something horribly wrong that I had to abort which resulted in a state without a working kernel. So i booted up with a live image to save data and reinstalled the os. Sorry @parkerlreed

z3ntu, 2019-07-28:
So can we close this?

Stefan538871, 2019-07-28:
I think so

closed 2019-07-28 by z3ntu