openrazer/openrazer #886: Razer Keypad Profile Support

linuxhelper, 2019-07-29:
Hi, I just noticed the daemon was updated a few hours ago. I'm wondering if the new release supports key-mapping or profiles.

My Nostromo keypad was used for keyboard shortcuts in Paint Tool Sai under Windows and I would like to be able to use it for Krita and other painting programs under Linux. My workflow suffers without it.

None of the OpenRazer applications seemed to support this feature previously. Keyboarding Master is dated and difficult to install and use (for a non-technical user).

One of the OpenRazer application pages previously stated that profiles were daemon dependent.
Is there another way to assign buttons (ctrl+z for example)?

If I'm mistaken, what can I use for my device?

Thanks :)

Example of button assignment and application profiles under Razer Configurator:
https://i.ytimg.com/vi/JF5xTc5LtHU/maxresdefault.jpg

z3ntu, 2019-07-29:
The PR #880 should hopefully work for you, could you test that? If you need any help in setting it up, just ask.

ZiggyInc, 2019-08-25:
OK, I too have this issue of not being able to map my Razer Orbweaver. could you tell me how to set it up? Linux mint 19.2 Openrazer and Polychromatic installed and working.

untamedeuphoria, 2019-08-27:
also running into the same issue with the older orbweaver. I cannot edit maccos. Open razor says that it has full support, but that support is not passed through to your app on my end.

closed 2020-01-24 by stale[bot]