openrazer/openrazer #891: ubuntu 19.04 Ornata Chroma "not signed with a trusted key"

JonasGroenbek, 2019-08-06:
The Polychromatic Controller GUI gives following error-message: 
A Razer product is connected to your computer, but the daemon didn't detect it.
It is possible your Razer product is not bound to the driver properly or it is not yet supported.

My Ornata Chroma has following PID ``1532:021E`` listed to be supported on the site

Kernel module log: ``dmesg``

``` 
[  779.389919] usb 1-1.1: new high-speed USB device number 11 using xhci_hcd
[  779.490268] usb 1-1.1: New USB device found, idVendor=1a40, idProduct=0101, bcdDevice= 1.00
[  779.490277] usb 1-1.1: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[  779.490282] usb 1-1.1: Product: USB2.0 HUB
[  779.491740] hub 1-1.1:1.0: USB hub found
[  779.491827] hub 1-1.1:1.0: 4 ports detected
[  779.877909] usb 1-1.1.1: new full-speed USB device number 12 using xhci_hcd
[  780.081791] usb 1-1.1.1: New USB device found, idVendor=1ea7, idProduct=0011, bcdDevice= 2.00
[  780.081802] usb 1-1.1.1: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[  780.081807] usb 1-1.1.1: Product: USB Mouse
[  780.090591] input: USB Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1.1/1-1.1.1/1-1.1.1:1.0/0003:1EA7:0011.0010/input/input45
[  780.150091] hid-generic 0003:1EA7:0011.0010: input,hidraw1: USB HID v1.10 Mouse [USB Mouse] on usb-0000:00:14.0-1.1.1/input0
[  780.151827] input: USB Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1.1/1-1.1.1/1-1.1.1:1.1/0003:1EA7:0011.0011/input/input46
[  780.210473] hid-generic 0003:1EA7:0011.0011: input,hidraw2: USB HID v1.10 Device [USB Mouse] on usb-0000:00:14.0-1.1.1/input1
[  780.389905] usb 1-1.1.3: new full-speed USB device number 13 using xhci_hcd
[  780.591824] usb 1-1.1.3: New USB device found, idVendor=1532, idProduct=021e, bcdDevice= 2.00
[  780.591833] usb 1-1.1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  780.591838] usb 1-1.1.3: Product: Razer Ornata Chroma
[  780.591842] usb 1-1.1.3: Manufacturer: Razer
[  780.601269] input: Razer Razer Ornata Chroma as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1.1/1-1.1.3/1-1.1.3:1.0/0003:1532:021E.0012/input/input47
[  780.659087] hid-generic 0003:1532:021E.0012: input,hidraw3: USB HID v1.11 Keyboard [Razer Razer Ornata Chroma] on usb-0000:00:14.0-1.1.3/input0
[  780.661537] input: Razer Razer Ornata Chroma Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1.1/1-1.1.3/1-1.1.3:1.1/0003:1532:021E.0013/input/input48
[  780.723055] input: Razer Razer Ornata Chroma Consumer Control as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1.1/1-1.1.3/1-1.1.3:1.1/0003:1532:021E.0013/input/input49
[  780.723593] input: Razer Razer Ornata Chroma System Control as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1.1/1-1.1.3/1-1.1.3:1.1/0003:1532:021E.0013/input/input50
[  780.724087] input: Razer Razer Ornata Chroma as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1.1/1-1.1.3/1-1.1.3:1.1/0003:1532:021E.0013/input/input51
[  780.724621] hid-generic 0003:1532:021E.0013: input,hidraw4: USB HID v1.11 Keyboard [Razer Razer Ornata Chroma] on usb-0000:00:14.0-1.1.3/input1
[  780.726907] input: Razer Razer Ornata Chroma as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1.1/1-1.1.3/1-1.1.3:1.2/0003:1532:021E.0014/input/input52
[  780.727607] hid-generic 0003:1532:021E.0014: input,hidraw5: USB HID v1.11 Mouse [Razer Razer Ornata Chroma] on usb-0000:00:14.0-1.1.3/input2
[  780.873052] PKCS#7 signature not signed with a trusted key
[  780.907977] PKCS#7 signature not signed with a trusted key
[  780.910710] PKCS#7 signature not signed with a trusted key
[  780.946910] PKCS#7 signature not signed with a trusted key
[  780.964272] PKCS#7 signature not signed with a trusted key
[  780.966179] PKCS#7 signature not signed with a trusted key

``` 

razer_mount log: ``sudo journalctl -b | grep -i razer_mount``
``` 
Aug 06 21:12:40 Jonas razer_mount[596]: Driver razerkbd
Aug 06 21:12:40 Jonas razer_mount[597]: Device_ID 0003:1532:021E.0005
Aug 06 21:12:40 Jonas razer_mount[599]: Modprobing razerkbd
Aug 06 21:12:40 Jonas systemd-udevd[434]: Process 'razer_mount razerkbd 0003:1532:021E.0005' failed with exit code 1.
Aug 06 21:12:40 Jonas razer_mount[603]: Driver razerkbd
Aug 06 21:12:40 Jonas razer_mount[604]: Device_ID 0003:1532:021E.0006
Aug 06 21:12:40 Jonas razer_mount[605]: Modprobing razerkbd
Aug 06 21:12:40 Jonas systemd-udevd[368]: Process 'razer_mount razerkbd 0003:1532:021E.0006' failed with exit code 1.
Aug 06 21:12:40 Jonas razer_mount[614]: Driver razerkbd
Aug 06 21:12:40 Jonas razer_mount[615]: Device_ID 0003:1532:021E.0004
Aug 06 21:12:40 Jonas razer_mount[621]: Modprobing razerkbd
Aug 06 21:12:40 Jonas systemd-udevd[376]: Process 'razer_mount razerkbd 0003:1532:021E.0004' failed with exit code 1.
Aug 06 21:14:22 Jonas razer_mount[4333]: Driver razerkbd
Aug 06 21:14:22 Jonas razer_mount[4334]: Device_ID 0003:1532:021E.0007
Aug 06 21:14:22 Jonas razer_mount[4336]: Modprobing razerkbd
Aug 06 21:14:22 Jonas systemd-udevd[4309]: Process 'razer_mount razerkbd 0003:1532:021E.0007' failed with exit code 1.
Aug 06 21:14:22 Jonas razer_mount[4348]: Driver razerkbd
Aug 06 21:14:22 Jonas razer_mount[4349]: Driver razerkbd
Aug 06 21:14:22 Jonas razer_mount[4350]: Device_ID 0003:1532:021E.0009
Aug 06 21:14:22 Jonas razer_mount[4351]: Device_ID 0003:1532:021E.0008
Aug 06 21:14:22 Jonas razer_mount[4352]: Modprobing razerkbd
Aug 06 21:14:22 Jonas razer_mount[4353]: Modprobing razerkbd
Aug 06 21:14:22 Jonas systemd-udevd[4307]: Process 'razer_mount razerkbd 0003:1532:021E.0008' failed with exit code 1.
Aug 06 21:14:22 Jonas systemd-udevd[4340]: Process 'razer_mount razerkbd 0003:1532:021E.0009' failed with exit code 1.
Aug 06 21:22:23 Jonas razer_mount[18706]: Driver razerkbd
Aug 06 21:22:23 Jonas razer_mount[18709]: Device_ID 0003:1532:021E.000B
Aug 06 21:22:23 Jonas razer_mount[18710]: Modprobing razerkbd
Aug 06 21:22:23 Jonas systemd-udevd[18698]: Process 'razer_mount razerkbd 0003:1532:021E.000B' failed with exit code 1.
Aug 06 21:22:23 Jonas razer_mount[18720]: Driver razerkbd
Aug 06 21:22:23 Jonas razer_mount[18722]: Driver razerkbd
Aug 06 21:22:23 Jonas razer_mount[18723]: Device_ID 0003:1532:021E.000A
Aug 06 21:22:23 Jonas razer_mount[18724]: Device_ID 0003:1532:021E.000C
Aug 06 21:22:23 Jonas razer_mount[18725]: Modprobing razerkbd
Aug 06 21:22:23 Jonas razer_mount[18726]: Modprobing razerkbd
Aug 06 21:22:23 Jonas systemd-udevd[18693]: Process 'razer_mount razerkbd 0003:1532:021E.000A' failed with exit code 1.
Aug 06 21:22:23 Jonas systemd-udevd[18699]: Process 'razer_mount razerkbd 0003:1532:021E.000C' failed with exit code 1.
Aug 06 21:23:32 Jonas razer_mount[18894]: Driver razerkbd
Aug 06 21:23:32 Jonas razer_mount[18898]: Device_ID 0003:1532:021E.000E
Aug 06 21:23:32 Jonas razer_mount[18901]: Modprobing razerkbd
Aug 06 21:23:32 Jonas razer_mount[18900]: Driver razerkbd
Aug 06 21:23:32 Jonas razer_mount[18903]: Driver razerkbd
Aug 06 21:23:32 Jonas razer_mount[18906]: Device_ID 0003:1532:021E.000D
Aug 06 21:23:32 Jonas systemd-udevd[18886]: Process 'razer_mount razerkbd 0003:1532:021E.000E' failed with exit code 1.
Aug 06 21:23:32 Jonas razer_mount[18905]: Device_ID 0003:1532:021E.000F
Aug 06 21:23:32 Jonas razer_mount[18911]: Modprobing razerkbd
Aug 06 21:23:32 Jonas razer_mount[18915]: Modprobing razerkbd
Aug 06 21:23:32 Jonas systemd-udevd[18887]: Process 'razer_mount razerkbd 0003:1532:021E.000F' failed with exit code 1.
Aug 06 21:23:32 Jonas systemd-udevd[18881]: Process 'razer_mount razerkbd 0003:1532:021E.000D' failed with exit code 1.
Aug 06 21:25:35 Jonas razer_mount[19196]: Driver razerkbd
Aug 06 21:25:35 Jonas razer_mount[19197]: Device_ID 0003:1532:021E.0013
Aug 06 21:25:35 Jonas razer_mount[19198]: Modprobing razerkbd
Aug 06 21:25:35 Jonas razer_mount[19205]: Driver razerkbd
Aug 06 21:25:35 Jonas systemd-udevd[19181]: Process 'razer_mount razerkbd 0003:1532:021E.0013' failed with exit code 1.
Aug 06 21:25:35 Jonas razer_mount[19212]: Driver razerkbd
Aug 06 21:25:35 Jonas razer_mount[19206]: Device_ID 0003:1532:021E.0012
Aug 06 21:25:35 Jonas razer_mount[19213]: Device_ID 0003:1532:021E.0014
Aug 06 21:25:35 Jonas razer_mount[19214]: Modprobing razerkbd
Aug 06 21:25:35 Jonas razer_mount[19215]: Modprobing razerkbd
Aug 06 21:25:35 Jonas systemd-udevd[19182]: Process 'razer_mount razerkbd 0003:1532:021E.0012' failed with exit code 1.
Aug 06 21:25:35 Jonas systemd-udevd[19173]: Process 'razer_mount razerkbd 0003:1532:021E.0014' failed with exit code 1.
``` 

I am not quite sure what to expect, or where to go from here. I would really love for this to work!

lah7, 2019-08-06:
It's likely your system has secure boot enabled and is refusing to load an unsigned driver. Without the driver, the daemon and frontends cannot use the devices.

Take a look at [Kernel modules on the Troubleshooting wiki page](https://github.com/openrazer/openrazer/wiki/Troubleshooting#kernel-modules-loading):

> The easiest solution is disabling Secure Boot in the BIOS/UEFI. The complicated solution is signing the built kernel modules, so that the kernel accepts them. See #98 or just use your favorite search engine.

JonasGroenbek, 2019-08-14:
It solved my question, thanks :)

closed 2019-08-18 by z3ntu