openrazer/openrazer #892: Fan Control for Razer Blade

DerMuffin, 2019-08-08:
### Distribution

Elementary OS 5.0

### Support

- [x ] Have you tried rebooting?
- [x ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Does this driver support the fan Control from synapse on Windows? Any Front end i find and install have "only" the options for makros and the Chroma for the Keyboard

z3ntu, 2019-08-08:
If you can make [USB packet captures](https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol) of Synapse doing that, I can look at it.

closed 2020-01-05 by stale[bot]

BiatuAutMiahn, 2023-06-19:
[RazerUSBCap.zip](https://github.com/openrazer/openrazer/files/11790239/RazerUSBCap.zip)
Compressed with ultra preset on 7-zip. This is my SKU: RZ09-0409CEA3-R3U1

z3ntu, 2023-07-16:
Please continue in #1227 or somewhere.