openrazer/openrazer #897: Lancehead wirelss mouse issue

wwwjunkboy, 2019-08-29:
● openrazer-daemon.service - Daemon to manage razer devices in userspace
   Loaded: loaded (/usr/lib/systemd/user/openrazer-daemon.service; enabled; vendor preset: enabled)
   Active: inactive (dead) since Wed 2019-08-28 18:59:24 PDT; 45ms ago
     Docs: man:openrazer-daemon(8)
  Process: 13886 ExecStart=/usr/bin/openrazer-daemon -Fv (code=exited, status=0/SUCCESS)
 Main PID: 13886 (code=exited, status=0/SUCCESS)

Aug 28 18:59:24 pop-os openrazer-daemon[13886]: AttributeError: 'RazerLanceheadWireless' object has no attribute 'getBattery'
Aug 28 18:59:24 pop-os openrazer-daemon[13886]: Exception ignored in: <function BatteryManager.__del__ at 0x7f2896ddaa60>
Aug 28 18:59:24 pop-os openrazer-daemon[13886]: Traceback (most recent call last):
Aug 28 18:59:24 pop-os openrazer-daemon[13886]:   File "/usr/lib/python3/dist-packages/openrazer_daemon/misc/battery_notifier.py", line 136, in __del__
Aug 28 18:59:24 pop-os openrazer-daemon[13886]:     self.close()
Aug 28 18:59:24 pop-os openrazer-daemon[13886]:   File "/usr/lib/python3/dist-packages/openrazer_daemon/misc/battery_notifier.py", line 126, in close
Aug 28 18:59:24 pop-os openrazer-daemon[13886]:     if not self._is_closed:
Aug 28 18:59:24 pop-os openrazer-daemon[13886]: AttributeError: 'BatteryManager' object has no attribute '_is_closed'
Aug 28 18:59:24 pop-os openrazer-daemon[13886]: Stopping daemon.
Aug 28 18:59:24 pop-os systemd[1583]: openrazer-daemon.service: Succeeded.

nightsky30, 2019-09-06:
@wwwjunkboy Can you please test the latest changes in #887?

z3ntu, 2019-09-24:
Closing as this is an issue only happening with a WIP pull request.

closed 2019-09-24 by z3ntu