openrazer/openrazer #898: KDE Neon Support

MuhammetDilmac, 2019-08-30:
Hello guys,
I try to add ppa for KDE Neon but not succeed. Can KDE Neon support be added?

```shell script
$ lsb_release -a
Distributor ID: neon
Description:    KDE neon User Edition 5.16
Release:        18.04
Codename:       bionic
```

z3ntu, 2019-09-24:
Hi, can you send me the error message you got? It should generally work with the PPA as KDE Neon is based on Ubuntu.

z3ntu, 2019-09-24:
It looks like a [bug for this issue](https://phabricator.kde.org/T9294) was closed on July 31st 2018, so I'm not sure why you're hitting that issue. But anyways, you should be able to get help in KDE Neon support channels, as this is definitely an issue on their side. I'm closing this issue but if you need anything else, feel free to comment.

closed 2019-09-24 by z3ntu