openrazer/openrazer #902: Starlight only works in monochromatic mode w/default color

jmachen, 2019-09-02:
Device is an original 2016 Razer Blade Stealth.  I believe my Black Widow Chroma v2 is exhibiting the same behavior, but I can't check right now.

### Distribution

5.2.8-1-MANJARO #1 SMP PREEMPT Fri Aug 9 19:34:20 UTC 2019 x86_64 GNU/Linux

### Support
I have tried rebooting.
I have looked at the troubleshooting wiki page

I am unable to activate Starlight with any color other than the default "Razer green" regardless of mode selection.  This works with single color, dual color, and random in Windows.  Here's the ouptut from dmesg.  Some of the weirder dmesg entries are me sending arbitrary values at the driver in hopes of stumbling upon a correct one (after a possible firmware update?).

> 
> [ 1667.594605] razerkbd: loading out-of-tree module taints kernel.
> [ 1667.594663] razerkbd: module verification failed: signature and/or required key missing - tainting kernel
> [ 1667.710775] razerkbd 0003:1532:0205.0005: input,hidraw3: USB HID v1.11 Keyboard [Razer Razer Blade Stealth] on usb-0000:00:14.0-8/input0
> [ 1667.897449] razerkbd 0003:1532:0205.0006: input,hidraw4: USB HID v1.11 Keyboard [Razer Razer Blade Stealth] on usb-0000:00:14.0-8/input1
> [ 1667.935736] razerkbd 0003:1532:0205.0007: input,hidraw5: USB HID v1.11 Mouse [Razer Razer Blade Stealth] on usb-0000:00:14.0-8/input2
> [ 1858.681419] razerkbd: Command failed. Start Marker: 03 id: ff Num Params: 02 Reserved: 0e Command: 84 Params: 01000000000000000000000000000000 .
> [ 1858.684223] razerkbd: Command failed. Start Marker: 03 id: ff Num Params: 03 Reserved: 03 Command: 80 Params: 01040000000000000000000000000000 .
> [ 2228.646566] razerkbd: Command failed. Start Marker: 03 id: ff Num Params: 01 Reserved: 03 Command: 0a Params: 19010100ff0000000000000000000000 .
> [ 2352.609388] razerkbd: Command failed. Start Marker: 03 id: ff Num Params: 01 Reserved: 03 Command: 0a Params: 19010100ff0000000000000000000000 .

jmachen, 2019-09-02:
In case it's helpful:

> Bus 001 Device 005: ID 1532:0205 Razer USA, Ltd Blade Stealth

lah7, 2019-09-02:
> 1667.594663] razerkbd: module verification failed: signature and/or required key missing - tainting kernel

I believe that's because the driver hasn't loaded because of secure boot. Try turning it off in the BIOS settings (easiest) or sign the driver yourself (bit more involved).

See here for further details:
https://github.com/openrazer/openrazer/wiki/Troubleshooting#kernel-modules-loading

jmachen, 2019-09-02:
> I believe that's because the driver hasn't loaded because of secure boot. Try turning it off in the BIOS settings (easiest) or sign the driver yourself (bit more involved).
> 
> See here for further details:
> https://github.com/openrazer/openrazer/wiki/Troubleshooting#kernel-modules-loading

Thanks @lah7 but it's already disabled and that particular error doesn't seem to have any bearing on the issue.  I am otherwise able to toggle between modes I just cannot specify the color for starlight, enable dual color mode for starlight, or enable random mode for starlight.

closed 2019-09-09 by z3ntu

jmachen, 2019-11-15:
This issue still persists.  Both dual and random modes still render in monochromatic mode with the default color as of the latest release.

jmachen, 2019-11-15:
I will rebuild tonight and try again to be doubly sure.

jmachen, 2019-11-19:
This is still an issue.  All three starlight modes are monochromatic. @z3ntu

nightsky30, 2019-11-20:
@jmachen Looked into it because I was curious.  #813 added references to the set_starlight_*_effect methods in keyboards.py, and the entries to create and remove the starlight effect device files in razerkbd_driver.c.  But case statements for Blade models weren't added in razerkbd_driver.c under razer_attr_write_mode_starlight.  The default case statement is executed which only triggers starlight single mode.

https://github.com/openrazer/openrazer/blob/974cefaac71388ea0c185df5fdf5c093b5651cc4/driver/razerkbd_driver.c#L1097

Adding a case statement for Blades which call the following will likely fix the issue:
razer_chroma_standard_matrix_effect_starlight_single
razer_chroma_standard_matrix_effect_starlight_dual
razer_chroma_standard_matrix_effect_starlight_random

jmachen, 2019-11-20:
> @jmachen Looked into it because I was curious. #813 added references to the set_starlight_*_effect methods in keyboards.py, and the entries to create and remove the starlight effect device files in razerkbd_driver.c. But case statements for Blade models weren't added in razerkbd_driver.c under razer_attr_write_mode_starlight. The default case statement is executed which only writes/triggers starlight single mode.
> 
> https://github.com/openrazer/openrazer/blob/974cefaac71388ea0c185df5fdf5c093b5651cc4/driver/razerkbd_driver.c#L1097
> 
> Adding a case statement for Blades which call the following will likely fix the issue:
> 
> razer_chroma_standard_matrix_effect_starlight_single
> razer_chroma_standard_matrix_effect_starlight_dual
> razer_chroma_standard_matrix_effect_starlight_random
> 
> OR adding the calls to handle all three modes in the default statement as handled similarly above that, and then creating a unique case for the BW2016 that only handles single mode...

I'm sure you're right, and I have made those changes, but I'm not sure how to correctly install the DKMS driver rebuilding.  So far I've done the following:

make clean
make driver_clean
make
sudo make driver_install

nightsky30, 2019-11-20:
I think you have to 'make all' before 'sudo make driver_install'

PR #907 (2019-09-09): Fix starlight single and dual effects
I noticed starlight single and dual effects didn't function properly on my keyboard (Razer Blackwidow Chroma V2) a while [back](https://github.com/openrazer/openrazer/pull/813#issuecomment-496720208), but hadn't gotten around to submitting the PR.  This should fix #902.

This PR fixes the starlight single and dual effects for the Razer Blackwidow Chroma V2 (and other starlight capable devices).

Prior to this PR:

Starlight single and dual effects only rendered monochrome lights (red in my case).

With these changes:

Starlight single and dual effects rend [... cut]