openrazer/openrazer #903: Support for Lian Li O11 Dynamic Razer Edition

tracker1, 2019-09-02:
### Device Info

* Name: Lian Li O11 Dynamic Razer Edition
* Product Number: us 001 Device 004: ID 1532:0f13 Razer USA, Ltd
* Model Number: 

### Device Information
```
Please post the output of
    lsusb -d 1532: -v
-----
Bus 001 Device 004: ID 1532:0f13 Razer USA, Ltd 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0f13 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
```

```
Please post the output of
    ls -lah /dev/input/by-id/
----
total 0
drwxr-xr-x 2 root root 220 Sep  2 03:34 .
drwxr-xr-x 4 root root 620 Sep  2 03:16 ..
lrwxrwxrwx 1 root root  10 Sep  2 03:34 usb-046d_C922_Pro_Stream_Webcam_40EE3E8F-event-if00 -> ../event23
lrwxrwxrwx 1 root root   9 Sep  2 03:34 usb-Logitech_USB_Laser_Mouse-event-mouse -> ../event2
lrwxrwxrwx 1 root root   9 Sep  2 03:34 usb-Logitech_USB_Laser_Mouse-mouse -> ../mouse0
lrwxrwxrwx 1 root root   9 Sep  2 03:34 usb-Razer_Lian_Li_O11_Dynamic_Designed_by_Razer-event-if01 -> ../event7
lrwxrwxrwx 1 root root   9 Sep  2 03:34 usb-Razer_Lian_Li_O11_Dynamic_Designed_by_Razer-event-kbd -> ../event3
lrwxrwxrwx 1 root root   9 Sep  2 03:34 usb-Razer_Lian_Li_O11_Dynamic_Designed_by_Razer-if01-event-kbd -> ../event4
lrwxrwxrwx 1 root root   9 Sep  2 03:34 usb-Razer_Lian_Li_O11_Dynamic_Designed_by_Razer-if02-event-mouse -> ../event8
lrwxrwxrwx 1 root root   9 Sep  2 03:34 usb-Razer_Lian_Li_O11_Dynamic_Designed_by_Razer-if02-mouse -> ../mouse1
lrwxrwxrwx 1 root root   9 Sep  2 03:34 usb-Unicomp_Inc_Ruffian6_x_Kbrd_v3_xx-event-kbd -> ../event9
```

### Packet Captures
<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
Don't have Windows Installed, only Linux -- would Windows in a VM work?

tracker1, 2019-09-02:
Would like to be able to help more with this than just create an issue/request...   I'm guessing I need to add a cfg to the virtual drivers directory, but not sure what I would need to change.  I'm also relatively sure that this device is probably the same as the chroma devkit with a different ID all things considered.

Not really familiar with python or this project, can only guess that the virtual drivers directory needs an additional .cfg, but unsure what I might need beyond this, how to build/deploy etc.

Did manage to build/install via `sudo make install_i_know_what_i_am_doing` even if I really don't... also installed Polychromatic, not sure if there's a better gui to be working with.... will add more if I am able to make progress... :-(

tracker1, 2019-09-02:
@z3ntu will Windows in a VirtualBox VM work?

z3ntu, 2019-09-02:
Yes, a Windows VM works fine. Our wiki page should tell you how to do the capture: https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol

tracker1, 2020-01-07:
Still interested in this, just haven't gotten a chance to get a windows VM or desktop working... I have to teardown later this week to upgrade my CPU, so was going to get another drive for windows boot in at the same time.

Will try and get the extra data captured over the weekend if I'm able.

tracker1, 2020-03-12:
Finally get around to setting up a Windows VM, and cannot get it working... I'm running an RC kernel (x570 mb, rx 5700 xt video card), and the kernel module won't compile.  I have actual windows on this machine, any chance that could work?

Looks like there's a USBpcap in the install options.. .I installed it and will try and get a good capture later today.  Any advice/help would be appreciated, will slog through it later.

Side note, can't believe Razer didn't split the control to a service, with the UI just for config, so that the settings stick when the screen is locked (last user on/in razor wins/sticks).

Ryzzack, 2020-06-28:
Hi,
Finally, did you find a way to make it work ? I also have Lian Lian Dynamic Razer Edition and I would be really happy if I find a way to control the RGB. I use opensuse. So far, my keyboard and mouse work fine with open Razer. But unfortunately; the case is not on it.
Thank you,

erickmathews, 2020-10-23:
Here are a few screens of the options for this case. It behaves much like a keyboard with individual LEDs along the front and bottom of the case.

[image]
[image]

StanTh3Man, 2020-11-14:
Hey I have this Case, and would love to see support for it! 

I have a working install of both fedora, and Windows 10.  

Do you guys have all the info that you need or can I provide anything?

I am a developer myself, but unfortunately I don't currently have time to spend implementing support for this myself :cry:

z3ntu, 2020-11-17:
@StanTh3Man USB Packet captures of Synapse controlling the lighting would help, see https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol

apocolipse, 2020-12-30:
I've got this case as well, went ahead and did a pcap (windows based w/ Wireshark and USBpcap).  I can do more to get some more info for setting different LED addresses and modes if needed but hoping it looks similar enough to other razer devices to work (fingers crossed). (File renamed to .txt so github would allow it)
[lianli-razer.txt](https://github.com/openrazer/openrazer/files/5752686/lianli-razer.txt)

z3ntu, 2021-01-02:
@apocolipse Did you have another Razer device connected while capturing the pcaps?

I see [this](https://github.com/openrazer/openrazer/blob/7c07ba0f7bd9ddcc5a47b818d3bbc0bfece3c292/driver/razeraccessory_driver.c#L399) used for setting the effect to custom frame and then a bunch of [these](https://github.com/openrazer/openrazer/blob/7c07ba0f7bd9ddcc5a47b818d3bbc0bfece3c292/driver/razeraccessory_driver.c#L681) calls with matrix dimensions [4, 16] but also more of [this](https://github.com/openrazer/openrazer/blob/7c07ba0f7bd9ddcc5a47b818d3bbc0bfece3c292/driver/razeraccessory_driver.c#L666) with matrix dimensions [6, 22]. And everything uses transaction id 0x1f

dweigert, 2021-01-23:
I'll be happy to test things if needed.  I have this case under Linux and the same case under Win10, Both machines have Razer keyboards and mice as well.  The Windows one has the Chroma HDK set up to provide under desk lighting, and I'm considering getting another one to do the same for the desk my Linux box is on.  The cooling for both machines is with a Thermaltake AIO for Threadripper processors, and I'll be working with the OpenLED folks to get that debugged and working on my Linux box as well.

LandonGetz, 2021-01-31:
Not sure where this is at, but I am willing to help/test as well....I also have the case in linux and win10. Would love to have my lighting setup completely working under linux...maybe i'll be able to remove that Win10 partition sometime soon :)

MajorEncore, 2021-03-22:
Sadly i lack this perticular skillset, but i have the case and a Asus Prime Z490-P mobo, the annoying part is that the case has shown up in synapse and worked fine when i built the pc, but only for about a day.. my guess is that it breaks due to some rando update..

dottat, 2021-11-05:
Physically it's definitely 3 strips. It seems like everything sorta just flows through as it's setup now rather than having different areas addressable.

ronoaldo, 2022-01-03:
I also have this case. Is there a way for me to help with testing/reverse-engineering/coding/debugging? Thanks!

dottat, 2022-01-03:
> I also have this case. Is there a way for me to help with testing/reverse-engineering/coding/debugging? Thanks!

Openrgb supports this one now. So the code is there.

ronoaldo, 2022-01-03:
Interesting. This is all new to me but reading the OpenRGB wiki, it says that it uses OpenRazer. It failed to detect mine (using flatpak).

dottat, 2022-01-03:
Leave open razor disabled in openrgb. I help get this one detected and it's
been working fine for months now. Run as admin first time only.

BigLadDylan, 2023-04-13:
Is there any update for the support of the chassis? I would really like to see my chassis' rgb aswell as my other devices!

dottat, 2023-04-13:
> Is there any update for the support of the chassis? I would really like to see my chassis' rgb aswell as my other devices!

It's been working for a while?

z3ntu, 2023-04-22:
It's not supported in OpenRazer yet. If you want to take a shot at adding to support for it, I've written some things at https://github.com/openrazer/openrazer/issues/903#issuecomment-753453747

I3urningstar, 2024-12-12:
any new Infos?