openrazer/openrazer #908: max_dpi for orochi 2011

iamdey, 2019-09-07:
### Distribution

debian 10

### Support

- [x] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Razergenie always get 500 for max_dpi but should be 4000

I see it is correctly defined https://github.com/openrazer/openrazer/blob/master/daemon/openrazer_daemon/hardware/mouse.py#L1003
And I found this default value here: https://github.com/openrazer/openrazer/blob/master/daemon/openrazer_daemon/dbus_services/dbus_methods/mamba.py#L177

z3ntu, 2019-09-07:
Thanks, I've pushed a fix to git master. You can apply that manually to the file in your filesystem if you want it fixed before the next release.

closed 2019-09-07 by z3ntu

iamdey, 2019-09-07:
wow! I believed there was a bug when posting the issue, but no you are just too fast :) 

Unfortunately I wasn't able to build the package, I follow [instructions](https://github.com/openrazer/openrazer/wiki/Building-a-package#debian-9) but I don't know how to install py deps:

```
dpkg-checkbuilddeps: erreur: Unmet build dependencies: dh-python
```

z3ntu, 2019-09-07:
You have to run `apt install dh-python`, debuild isn't smart enough to install the dependencies itself :wink: 

But you can also just patch it locally without building the package by editing `/usr/lib/python3/dist-packages/openrazer_daemon/hardware/mouse.py` and applying the fix from https://github.com/openrazer/openrazer/commit/fc1d92e0a16e9fc882d824d969a38308f78225d9 manually; that's probably less work.

iamdey, 2019-09-07:
I did build the .deb (very easy finally), all works as expected \o/.

commit fc1d92e0 (2019-09-07): Fix max DPI for Orochi 2011

Fixes #908