openrazer/openrazer #909: why my mouse settings are forced ?

iamdey, 2019-09-07:
### Distribution

Debian 10

### Support

- [x] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

On my orochi 2011, I have factory settings and set those parameters:
* poll_rate is at 1000 (or 500, don't remember)
* dpi is at 2000
* btn 6 and 7 allow to change dpi (something like 125, 500, 1500, 2000 , 4000) and also it toggles microphone on linux

### Main question:
After installing the openrazer driver, why does the dpi set to 500 (I'm not sure if it is related to the max_dpi and razergenie bug) and why buttons 6 & 7 doesn't have effect anymore ?

How can I reset to factory setting without going on windows and using original razer software?

---

### Other questions for my understanding:

I saw here the mouse initialization, but I don't know how to understand it https://github.com/openrazer/openrazer/blob/master/driver/razermouse_driver.c#L2619-L2620

```
void razer_mouse_init(struct razer_mouse_device *dev, struct usb_interface *intf, struct hid_device *hdev)
// … 
// Setup orochi2011
dev->orochi2011_dpi = 0x4c;
dev->orochi2011_poll = 500;
```

I suppose the device is not written at this moment. 
What is the meaning of `0x4c` (I'm not fluent with code for hardware)

z3ntu, 2019-09-07:
> why does the dpi set to 500

From looking at the driver code, I don't see any location where the DPI gets set without the user asking explicitly.

> why buttons 6 & 7 doesn't have effect anymore

On newer devices it has something to do with the "driver mode" but the Orochi 2011 doesn't support that. Can you try stopping the openrazer daemon (`systemctl --user stop openrazer-daemon.service`) and see if the same behavior persists? If that doesn't change anything you can try uninstall openrazer completely from your system and rebooting and see if that changes anything.

iamdey, 2019-09-07:
> why does the dpi set to 500
>
> From looking at the driver code, I don't see any location where the DPI
> gets set without the user asking explicitly.
Yes I suppose it's me when I trier razergenie.

> why buttons 6 & 7 doesn't have effect anymore
>
> On newer devices it has something to do with the "driver mode" but the
> Orochi 2011 doesn't support that. Can you try stopping the openrazer daemon
> (systemctl --user stop openrazer-daemon.service) and see if the same
> behavior persists

OK I'll try that.

If that doesn't change anything you can try uninstall openrazer completely
> from your system and rebooting and see if that changes anything.
I did that. But nothing changed, and more, when I plugged in an other
computer, the mouse keep the settings I set with razergenie.

(As said above, I was able to reset factory on the official software )

closed 2020-02-04 by stale[bot]