openrazer/openrazer #910: Support for Razer Blade 17" (2019)

adamhill, 2019-09-08:
### Device Info

* Name: Razer Blade Pro 17 (2019)
* Product Number: RZ09-02878E92
* Model Number: R3U1

### Device Information
```
Please post the output of
    lsusb -d 1532: -v

Bus 001 Device 003: ID 1532:0234 Razer USA, Ltd 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0234 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Blade
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)
```
```
Please post the output of
    ls -lah /dev/input/by-id/

total 0
drwxr-xr-x 2 root root 180 Sep  7 22:48 .
drwxr-xr-x 4 root root 640 Sep  7 22:48 ..
lrwxrwxrwx 1 root root   9 Sep  7 22:48 usb-Azurewave_Integrated_Camera_200901010001-event-if00 -> ../event5
lrwxrwxrwx 1 root root   9 Sep  7 22:48 usb-Azurewave_Integrated_Camera_200901010001-event-if02 -> ../event6
lrwxrwxrwx 1 root root  10 Sep  7 22:48 usb-Razer_Razer_Blade-event-if01 -> ../event11
lrwxrwxrwx 1 root root   9 Sep  7 22:48 usb-Razer_Razer_Blade-event-kbd -> ../event7
lrwxrwxrwx 1 root root   9 Sep  7 22:48 usb-Razer_Razer_Blade-if01-event-kbd -> ../event8
lrwxrwxrwx 1 root root  10 Sep  7 22:48 usb-Razer_Razer_Blade-if02-event-mouse -> ../event12
lrwxrwxrwx 1 root root   9 Sep  7 22:48 usb-Razer_Razer_Blade-if02-mouse -> ../mouse2
```
### Packet Captures
How do I go about packet capture for a keyboard in the laptop?

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
Working on it next.

<Screenshots of all available options as seen on Razer Synapse>

legomind, 2019-12-04:
Any chance of this happening soon?

z3ntu, 2019-12-15:
If you're willing to contribute, you can take a look at recent pull requests for Razer Blade laptops, they should be a good reference. As far as I am aware, most Blades have exactly the same functionality so it could/should be a copy-paste job.

z3ntu, 2020-05-02:
Fixed with #916

closed 2020-05-02 by z3ntu