openrazer/openrazer #917: Interfacing with the driver - question

Jacajack, 2019-09-17:
### Support

Hi,

I have a question about the low-level driver interface described on your [wiki](https://github.com/openrazer/openrazer/wiki/Using-the-keyboard-driver).
It says there that the page is probably outdated and only valid to certain extent, however the code snippets seem to work fine with my keyboard.

I've also found following statement on Polychromatic FAQ page:
> Currently, one of the OpenRazer team members is rewriting OpenRazer to handle everything in userspace, which will eliminate the dependency on DKMS drivers for GNU/Linux. This rewrite uses hidapi and is making excellent progress, but is still in early development.

Hence my question: is the current driver interface going to be replaced anytime soon?
If so, are you planning on exposing any low-level interface similar to the current one?
Also, is it still valid to use it directly in the first place?

Thank you in advance

z3ntu, 2019-09-24:
> It says there that the page is probably outdated and only valid to certain extent, however the code snippets seem to work fine with my keyboard.

The driver side doesn't change much but most people should use the dbus interface or just a GUI and not the driver directly, as you it's quite ugly to use and doesn't have a lot of safeguards.

> is the current driver interface going to be replaced anytime soon?

Hopefully :)

> If so, are you planning on exposing any low-level interface similar to the current one?

No, the new project just has a dbus interface for interaction

> Also, is it still valid to use it directly in the first place?

I'm not sure why one would want to use the driver interface instead of talking to the daemon over dbus (or use the python bindings accessing the dbus interface) but nobody will stop you :wink: 

Any further questions?

Jacajack, 2019-09-24:
First of all, thank you for the exhaustive answer! :)

> I'm not sure why one would want to use the driver interface instead of talking to the daemon over dbus (or use the python bindings accessing the dbus interface) but nobody will stop you wink

I was just considering writing a _very_ simple C daemon, mostly for fun.  Something that would take row and column numbers and RGB values from _stdin_, and light up some LEDs on my keyboard accordingly. I think it could become a pretty powerful tool with right scripting, too

Maybe I'm paranoid, but I just felt like two layers of Python and dbus between me and the keyboard would be an overkill for that :wink:

> No, the new project just has a dbus interface for interaction

Honestly, outside of Python, dbus doesn't seem to be very user friendly at all... I don't want to be intrusive, but have you considered using other simpler interfaces (UNIX sockets, FIFO files, etc.) as well?

z3ntu, 2019-09-25:
> Maybe I'm paranoid, but I just felt like two layers of Python and dbus between me and the keyboard would be an overkill for that wink

I can kind of understand it but honestly for a simple thing, using ten layers of abstraction is fine too :wink: 

> Honestly, outside of Python, dbus doesn't seem to be very user friendly at all...

The new daemon is written in C++ with Qt which works very nicely there. I've also already written Go bindings for the dbus interface and it works fine there too. And even for C there are a couple of libraries out there (which apparently are all quite bad except sd-bus).

> have you considered using other simpler interfaces (UNIX sockets, FIFO files, etc.) as well?

Not really but I know that dbus is much friendlier for structured data than those and is also more modern. And dbus works great for a lot of projects.

closed 2019-09-25 by z3ntu

Jacajack, 2019-09-26:
> I can kind of understand it but honestly for a simple thing, using ten layers of abstraction is fine too wink

Yeah, that's true as well :)

> The new daemon is written in C++ with Qt which works very nicely there. I've also already written Go bindings for the dbus interface and it works fine there too. And even for C there are a couple of [...]

> Not really but I know that dbus is much friendlier for structured data than those and is also more modern. And dbus works great for a lot of projects.

I get your point. Maybe it's high time for me to get myself more familiar with dbus then.

Thanks again, for the answer, and for the great project. :)
<hr>
By the way, I made the "daemon" I was talking about... only as an awk-abusing one-liner :wink:.
I thought I'd share:

```awk
BEGIN{fm=ENVIRON["DRIVER"]"/matrix_custom_frame";fe=ENVIRON["DRIVER"]"/matrix_effect_custom";}NF==5{for(i=3;i<6;i++)m[$1,$2,i-3]=$i;}/w/{for(y=0;y<6;y++){printf"%c%c%c",y,0,21>fm;for(x=0;x<22;x++)printf"%c%c%c",m[y,x,0]%256,m[y,x,1]%256,m[y,x,2]%256>fm;}fflush(fm);print"1">fe;fflush(fe);}
```

Run like this: `DRIVER=/sys/bus/hid/drivers/razerkbd/xxx LC_ALL=C awk -f razer.awk < data.txt`.
And here's some data:
```
3 2 255 0 0
3 3 255 128 0
3 4 255 255 0
3 5 128 255 0
3 6 0 255 0
3 7 0 255 128
3 8 0 255 255
3 9 0 128 255
3 10 0 0 255
3 11 128 0 255
3 12 255 0 255
w
```