openrazer/openrazer #918: Support for Razer Abyssus Essential

michalzimniak, 2019-09-22:
### Device Info

* Name: Razer Abyssus Essential
* Product Number: RZ01-Q2160300
* Model Number: RZ01-Q216

### Device Information
```
Bus 001 Device 003: ID 1532:006b Razer USA, Ltd 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x006b 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Abyssus Essential
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1

```

```
lrwxrwxrwx 1 root root   9 wrz 22 03:01 usb-Razer_Razer_Abyssus_Essential-event-if01 -> ../event7
lrwxrwxrwx 1 root root   9 wrz 22 03:01 usb-Razer_Razer_Abyssus_Essential-event-mouse -> ../event3
lrwxrwxrwx 1 root root   9 wrz 22 03:01 usb-Razer_Razer_Abyssus_Essential-if01-event-kbd -> ../event4
lrwxrwxrwx 1 root root   9 wrz 22 03:01 usb-Razer_Razer_Abyssus_Essential-if02-event-kbd -> ../event8
lrwxrwxrwx 1 root root   9 wrz 22 03:01 usb-Razer_Razer_Abyssus_Essential-mouse -> ../mouse0

```

z3ntu, 2019-09-24:
Duplicate of #612

z3ntu, 2020-01-05:
Re-opening because the author of #612 closed the issue (and this issue has nicer information)

Kern3l-Panic, 2020-02-20:
It's nice to see this re-opened and still being worked on. I hope to see support for this device soon :) You guys are doing good work here! Thank you.

commit db9486bc (2020-04-05): Add support for Razer Abyssus Essential

Fixes #918
closed 2020-04-05 by z3ntu