openrazer/openrazer #919: Issue with razerblade late 2016

KosenJPG, 2019-09-26:
### Distribution

Manjaro linux 5.1.12-21

### Support

- [x] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

I don't understand why openrazer is not working. I read everything on the troubleshooting page but nothing helps and it's starting to drive me crazy. I already added my user to the plugdev group and tried everything listed but whenever I try to `sudo modprobe razerkbd` or `sudo modprobe 1532:020f` it doesn't work and just outputs `modprobe: FATAL: Module 1532:020f not found in directory /lib/modules/5.1.21-1-MANJARO`. 

I also tried running `openrazer-daemon -Fv` but it just says `Unable to lock on the pidfile`. I also checked the logs but no errors appeared. Can someone help me please it's been months and I really want my keyboard to light up correctly.

z3ntu, 2019-09-26:
Do you have your kernel headers (`linux51-headers`) installed?

KosenJPG, 2019-09-26:
> Do you have your kernel headers (`linux51-headers`) installed?

It was actually that simple... I feel really dumb, thank you for your help!

z3ntu, 2019-09-26:
No problem!

closed 2019-09-26 by z3ntu