openrazer/openrazer #920: Set DPI of mouse at login

killerb77, 2019-10-02:
### Distribution

Linux Mint 19.2 Cinnamon
### Support

- [ ] Have you tried rebooting?
- [yes ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

What would be the string to add to my start up applications to set the DPI at login?

Like this one: python3 -c "import openrazer.client; a = openrazer.client.DeviceManager(); a.devices[0].fx.static(0, 255, 0) which I was able to find from another post and modify it to set the effect and color i wanted.

Thank you in advance.

tildearrow, 2019-10-03:
Should I also add support for this to #888?

z3ntu, 2019-10-03:
If you only have one device, then it's `a.devices[0].dpi = (500, 500)` as the last statement (for example, you can modify the DPI values of course). If you have multiple devices I would suggest you creating a loop that goes through all devices and checks if `device.type` is `mouse` or check `device.name` to be your mouse, as you obviously can't set DPI on a keyboard or on other devices.

killerb77, 2019-10-03:
@z3ntu I have only one device.  That worked.  Thank you!

z3ntu, 2019-10-03:
Great :)

closed 2019-10-03 by z3ntu