openrazer/openrazer #922: Not avaiable on Kali Linux?

fnagua, 2019-10-06:
### Distribution

### Support

- [ ] Have you tried rebooting?
- [ ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Hi, is it possible to install it on kali linux? I´ve tried to install all the devian programs but it really doesn´t work.
Thnx!

z3ntu, 2019-10-08:
Hi, from what I could find online, Kali Linux is based on Debian Testing, so the instructions on our website for Debian Testing should work. What errors are you getting?

fnagua, 2019-10-09:
Hi, because of the error of kali with gnome i change my distro to debian 10, and i have the same problem, i cant install it with no reason... maybe im so stupid or something but i really cant jajaja help pls

z3ntu, 2019-10-09:
Please post the errors you're getting, otherwise I can't help you.

closed 2020-02-08 by z3ntu