openrazer/openrazer #928: Support for Razer Blade 15 (2018 Advanced model)

GRX78FL, 2019-10-15:
### Device Info

* Name: Razer Blade 15 (2018 Advanced Model)
* Product Number: RX09-02385E92
* Model Number: RZ09-02385

### Device Information
```
    lsusb -d 1532: -v

   Bus 001 Device 005: ID 1532:0233 Razer USA, Ltd Razer Blade
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0233 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1

```

```
    ls -lah /dev/input/by-id/

    total 0
drwxr-xr-x. 2 root root 200 Oct 15 04:41 .
drwxr-xr-x. 4 root root 600 Oct 15 04:41 ..
lrwxrwxrwx. 1 root root  10 Oct 15 04:41 usb-11918920-008ADB02_USB_Camera-event-if00 -> ../event14
lrwxrwxrwx. 1 root root   9 Oct 15 04:41 usb-Logitech_USB_Receiver-if02-event-mouse -> ../event6
lrwxrwxrwx. 1 root root   9 Oct 15 04:41 usb-Logitech_USB_Receiver-if02-mouse -> ../mouse0
lrwxrwxrwx. 1 root root  10 Oct 15 04:41 usb-Razer_Razer_Blade-event-if01 -> ../event11
lrwxrwxrwx. 1 root root   9 Oct 15 04:41 usb-Razer_Razer_Blade-event-kbd -> ../event7
lrwxrwxrwx. 1 root root   9 Oct 15 04:41 usb-Razer_Razer_Blade-if01-event-kbd -> ../event8
lrwxrwxrwx. 1 root root  10 Oct 15 04:41 usb-Razer_Razer_Blade-if02-event-mouse -> ../event12
lrwxrwxrwx. 1 root root   9 Oct 15 04:41 usb-Razer_Razer_Blade-if02-mouse -> ../mouse1

   
```

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on Razer Synapse>
Will attach soon, I am currently unable to boot my Windows 10 installation

z3ntu, 2019-12-15:
Hi, the device is already supported since v2.4.0 so it should work fine out of the box? 

What issues do you see on your install?

GRX78FL, 2019-12-21:
This issue can be closed as my problems magically fixed themselves.
Thanks for the reply and for this project.

closed 2019-12-21 by GRX78FL