openrazer/openrazer #929: Support for [Razer Core Chroma]

Rems38, 2019-10-15:
### Device Info

* Name: CORE X CHROMA - External graphics enclosure
* Product Number: NO NUMBER
* Model Number: RC21-0143

### Device Information
```
lsusb -d 1532: -v

Bus 001 Device 003: ID 1532:0f1a Razer USA, Ltd 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0f1a 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1

```

```
Please post the output of
    lrwxrwxrwx 1 root root  10 oct.  15 13:32 usb-Razer_Razer_Core_X_Chroma-event-if01 -> ../event11
lrwxrwxrwx 1 root root   9 oct.  15 13:32 usb-Razer_Razer_Core_X_Chroma-event-kbd -> ../event7
lrwxrwxrwx 1 root root   9 oct.  15 13:32 usb-Razer_Razer_Core_X_Chroma-if01-event-kbd -> ../event8
lrwxrwxrwx 1 root root  10 oct.  15 13:32 usb-Razer_Razer_Core_X_Chroma-if02-event-mouse -> ../event12
lrwxrwxrwx 1 root root   9 oct.  15 13:32 usb-Razer_Razer_Core_X_Chroma-if02-mouse -> ../mouse2

```

### Packet Captures

### Synapse Screenshots
I don't have a windows computer :/

turbos-and-tacos, 2019-12-15:
Rename the extension to .pcap and WireShark should be able to open it.

[egpu_various_effects.txt](https://github.com/openrazer/openrazer/files/3965214/egpu_various_effects.txt)

z3ntu, 2019-12-15:
Thanks, the pcap is fine. Unfortunately Synapse is also just using the "SET EXTENDED CUSTOM FRAME" effects for this device and those aren't really implemented in openrazer yet.

turbos-and-tacos, 2019-12-15:
Anything I can do differently to get a better pcacp?

z3ntu, 2019-12-16:
Unfortunately not. The Razer Synapse 3 software just doesn't give us the information we would like to have.

devx97, 2020-06-06:
Can I help somehow? I have Razer Core X Chroma, too. And I really want to get this working together ;) Also, I have ASUS Card with AURA in it.

dasizeman, 2020-09-19:
Any progress here? Would like to help get support added if possible too. What is the blocker here, and do we understand anything about the protocol for these newer devices?

widget7h15, 2020-10-07:
I have the Razer Core X - Non Chroma edition and can help to gather data

redoxeon, 2021-12-06:
I have a Core X Chroma with a 1050Ti and would also love to help with this issue!

jershell, 2021-12-24:
Hello, any news?

turbos-and-tacos, 2021-12-28:
I have the device recognized by the driver. (populating serial numbers and such). Still working on getting it to accept lighting commands.

rosterloh, 2022-09-04:
[image]

z3ntu, 2023-01-09:
Please help test #1998 , see https://github.com/openrazer/openrazer/wiki/Building-a-package how to build from source.

jershell, 2023-01-10:
I have a core x chroma and ubuntu 22.04.1 LTS AMD64. 
i did
by this instruction https://github.com/openrazer/openrazer/wiki/Building-a-package
```sh
git clone https://github.com/turbos-and-tacos/openrazer.git -b core-x-support
cd openrazer
sudo apt install debhelper dh-python linux-headers-generic python3-setuptools lsb-release dkms # build dependencies
./scripts/build_debs.sh

sudo dpkg -i ./dist/*.deb
```
I have success build and install debs
In first i check a daemon output
openrazer-daemon -Fv
...
2023-01-10 21:00:22 | razer                          | INFO     | Found device.0: 0003:1532:0F1A.0005
2023-01-10 21:00:22 | razer.device0                  | INFO     | Initialising device.0 RazerCoreXChroma
...
OK. 

Next, i used a polychromatic with version 0.8.0 for test. 
1. Select a wave effect
[image]

https://user-images.githubusercontent.com/1238301/211631726-c902e4f4-28d4-4a93-b77d-e52f13f013ea.mp4

2. Select a static effect with 75% of brightness level
[image]
Its work.
[image]

3. Select a dual starlight 
 
[image]

https://user-images.githubusercontent.com/1238301/211632372-1ce17643-77e3-4415-9b02-d8f0dfb314dc.mp4

lgtm

Ugzuzg, 2023-01-10:
Works for me on Arch Linux, too.

It doesn't look like the wave effect goes in the right direction.

closed 2023-01-15 by z3ntu