openrazer/openrazer #930: Cant get it to work on solus

ItsLeffdePeff, 2019-10-16:
I tried so many things i tried using polychromatic normal commands razergenie and the troubleshoot page but nothing seems to work can somebody help me please

z3ntu, 2019-10-16:
Can you please describe your issue in more detail? What error do you see?

ItsLeffdePeff, 2019-10-16:
Razer device not detected by deamon

ItsLeffdePeff, 2019-10-16:
And with polychromatic razer device not recognised

parkerlreed, 2019-10-16:
Logs from the daemon? This isn't much information to go on

z3ntu, 2019-10-17:
@ItsLeffdePeff Can you give me the output of 
`sudo dkms install openrazer-driver/2.6.0` ?

ItsLeffdePeff, 2019-10-17:
sudo: dkms: command not found and modprobe: FATAL: Module razerkbd not found in directory /lib/modules/5.2.20-130.current

z3ntu, 2019-10-17:
Try installing the latest updates and reboot?

ItsLeffdePeff, 2019-10-18:
all the latest updates were already installed

z3ntu, 2019-10-18:
Then please ask at Solus for help. The latest version in Solus should provide `/lib/modules/5.2.20-130.current/kernel/drivers/hid/razerkbd.ko` which matches the kernel version from your output above.

Maybe one last shot, do you have the package `openrazer-current` installed?

ItsLeffdePeff, 2019-10-18:
openrazer-current worked!!! thank you so much

z3ntu, 2019-10-19:
Great!

closed 2019-10-19 by z3ntu