openrazer/openrazer #932: Recognizes Devices but Issue with Drivers

triplecombinationflusterbuster, 2019-10-22:
### Distribution

### Support

- [ ] Have you tried rebooting?
- [ ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

I'm currently trying to get my Razer Deathadder Elite and Razer Blackwidow Elite to communicate with Polychromatic.  Polychromatic states that my PC recognizes one (or both?) Razer products but the daemon is not detecting it.  I have searched high and low for a solution on the internet but all possible solutions that I found have turned up no dice.  

I will post some outputs here that I believe to be relevant considering past forum posts I have read but please be patient - I have a very elementary understanding of code.

`modprobe razerkbd`
`modprobe: FATAL: Module razerkbd not found in directory /lib/modules/3.14.0`

The files regarding the drivers just don't exist however I have installed everything properly and added my user to the plugdev group (and restarting!).

I came to believe that their might have been an issue with the kernel headers, here's the relevant information:
`uname -r`
`3.14.0`

Which I find weird as I found that running `sudo install openrazer-meta` installed kernel headers for me (the version was like 4.14.4 or something like that).  Also, I am on openrazer 2.6.0. 

Running `sudo apt install linux-headers-generic` outputs `linux-headers-generic is already the newest version (4.4.0.166.174).
The following packages were automatically installed and are no longer required:
  linux-headers-4.4.0-165 linux-headers-4.4.0-165-generic`

I am horribly confused, any help would be much appreciated and thank you in advance.  I am also willing to provide any further relevant information that I may have left out.

lah7, 2019-10-22:
Which version of Ubuntu (XFCE) are you running?

It looks like you've installed the driver for the **4.4.0** kernel, but your system is still running an older kernel - **3.14.0**.

If running this old kernel is an accident, hold <kbd>SHIFT</kbd> while booting the system, choose "Ubuntu (advanced options)" and pick the latest kernel. If that works, you may wish to remove older kernels and update the GRUB bootloader.

If there's a need to stick with this old kernel, install the kernel header package for that version - which I would guess is `linux-headers-generic-314` or similar.

triplecombinationflusterbuster, 2019-10-23:
> Which version of Ubuntu (XFCE) are you running?
> 
> It looks like you've installed the driver for the **4.4.0** kernel, but your system is still running an older kernel - **3.14.0**.
> 
> If running this old kernel is an accident, hold SHIFT while booting the system, choose "Ubuntu (advanced options)" and pick the latest kernel. If that works, you may wish to remove older kernels and update the GRUB bootloader.
> 
> If there's a need to stick with this old kernel, install the kernel header package for that version - which I would guess is `linux-headers-generic-314` or similar.

I've tried repeatedly to install the headers for 3.14.0 but I get this error output `E: Unable to locate package linux-headers-generic-3.14.0
E: Couldn't find any package by glob 'linux-headers-generic-3.14.0'
E: Couldn't find any package by regex 'linux-headers-generic-3.14.0'`

This basically tells me that my system is confused because I believe the 3.14.0 kernels were done away with a while ago so they don't show up in the list anymore.

lah7, 2019-10-23:
Which version is this environment running?

    lsb_release -a
    uname -a

The generic 4.4 kernel sounds most likely to be 16.04 (Xenial), which is still supported. Not being familiar with Crouton or Chromebooks, I'd assume you'll need to compile a custom 3.14 kernel and include OpenRazer's module somehow:

* https://github.com/dnschneid/crouton/wiki/Build-chrome-os-kernel-and-kernel-modules

**Or...** you have a headers package provided by Crouton, but the kernel itself isn't `generic`. Have a look in a GUI apt tool like **Synaptic Package Manager** as there could be a `linux-headers-` package that provides headers for this custom 3.14 kernel.

Chroots use the host system's kernel, and it's not possible for a chroot to use a different kernel. The chroot (Ubuntu) just happens to have a 4.4 kernel installed, but won't be used.

triplecombinationflusterbuster, 2019-10-24:
`lsb_release -a` outputs:
`No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.6 LTS
Release:	16.04
Codename:	xenial`

&

`uname -a` outputs:
`Linux localhost 3.14.0 #1 SMP PREEMPT Wed Sep 25 23:32:59 PDT 2019 x86_64 x86_64 x86_64 GNU/Linux`

I have to say I'm starting to get quite confused because I am on 16.04 yet it's hellbent on using 3.14.0 headers?  Given this information. there shouldn't be a problem at all - other than the fact that the openrazer drivers aren't present in the /usr/ folder (referenced from an older thread).

lah7, 2019-10-24:
Crouton (the chroot) inherits the kernel from the host system, which is ChromeOS (3.14.0). This answer helps explain the situation:

* https://unix.stackexchange.com/a/356636

VirtualBox is another piece of software that requires a DKMS module to run, so it has the same limitation of chroots.

It turns out it won't be possible to use DKMS modules within the chroot, since it doesn't have access to the host kernel, which is a ChromeOS kernel and not the generic one. **I'm not familiar with this**, but you'll need to build the module **on ChromeOS** for it to work inside the chroot:

* https://github.com/dnschneid/crouton/wiki/Build-chrome-os-kernel-and-kernel-modules

---

The alternate would be to dual boot with [GalliumOS](https://galliumos.org/), a Linux distro based on Ubuntu and XFCE that runs the 'generic' kernel.

* https://en.wikipedia.org/wiki/GalliumOS

If your Acer Chromebook 13 model is from 2018, [then it seems to be supported](https://wiki.galliumos.org/Hardware_Compatibility).

triplecombinationflusterbuster, 2019-10-24:
I see. Thank you so much for your informative help.  I will attempt to build the kernel module and see what happens - hell, it's just a chroot so if it goes belly up I can just pulverize it and start all over.  

If I could trouble for one more issue, which is probably a little off-topic but it does relate to my Razer peripherals, I have also been struggling with completely disabling mouse acceleration on my chroot.  I have reached a few solutions that seem to have worked such as adding a _90-mouse.conf_ file to my xorg.conf.d folder but multiple sources have different named files such as _20-mouse.conf_ and _50-mouse.conf_. No mouse.conf file existed prior, **I had to create it**.  Could you shed any light on this and maybe suggest a more surefire way to completely disable mouse acceleration?  I have been using `xinput` to modify the device acceleration profile (`xinput set-prop x 'Device Accel Profile' -1` but there's no true definitive confirmation that acceleration has been completely disabled versus, say, modifying mouse sensitivity settings on Windows.

lah7, 2019-10-24:
> Thank you so much for your informative help. I will attempt to build the kernel module, post my results, and see what happens - hell, it's just a chroot so if it goes belly up I can just pulverize it and start all over.

No problem - do bear in mind you'll be building the kernel **outside the chroot.** No idea how general Linux runs on Chromebooks since I haven't owned one. :)

> Could you shed any light on this and maybe suggest a more surefire way to completely disable mouse acceleration?

Outside of my knowledge, sorry. I guess you'll want to check XFCE mouse settings in case this is overriding what you've set in Xorg config files, or add the command to a script to run when the XFCE session starts.

z3ntu, 2019-12-15:
Closing as it's not an openrazer issue.

closed 2019-12-15 by z3ntu